realthunder/FreeCAD #349: Shortcut for Sketch Polyline tangent, perpendicular etc styles with "M" not working (Daily 2022.111)

Miksu80, 2022-04-15:
As description says, in sketch cycling polyline modes with "M" not working at all.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/Finland (en_FI)

raindropsfromsky, 2023-06-21:
I am still facing this issue in June 2023: While drawing the polyline, pressing M terminates the polyline.
I tried to press G,M but that does not change the mode.
If I press the M key twice rapidly, it does not terminate the polyline. But this (M,M) shortcut does not change the mode, either.

I am using realthunder/FreeCAD Linkbranch 2023.524.0.14555, Python 3.11.3, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1.
My OS is Windows 11 Home.

I checked the hotkey settings, and they seem to be fine. It defines only the M key, and not G,M or M,M.
[image]

dickshaydle, 2023-06-21:
Thank you. I also wanted to file that but.
I can reproduce that on Linux mint and Arch. With the latest release (Tip 2023.05.21) and already version 2023.05.16, but not 2023.01.31. 

'M' is supposed to cycle through the modes and not end the line.

This big was introduced

blue0ceanby, 2023-06-21:
as a workaround if you run it as shift M a switch is possible

dickshaydle, 2023-06-21:
Something that was easy before turns into a hidden feature with Shift+M.

It was intuitive and simpler before to press the same key as the polyline tool itself.

commit 2703b79f (2023-06-24): Sketcher: fix polyline command toggle

Fixes #349