realthunder/FreeCAD #356: [Issue] Chamfer not work

Kuzma30, 2022-04-16:
```
OS: Ubuntu Jammy Jellyfish (development branch) (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Ukrainian/Ukraine (uk_UA)
```
I try select Face or Edge for Chamfer. It doesn't work and show error 
15:57:21  63.102 <App> Document.cpp(4266): Failed to recompute testcase_FCSd#Chamfer: BRep_API: command not done
Then I create test file with simple Body with Pad and Pocket.
For Pad Chamfer working, for Pocket - none.

[image]

bambukouk, 2022-04-16:
[image]

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```
???

Kuzma30, 2022-04-16:
How to attach test file here?

bambukouk, 2022-04-16:
The difference is (I think) that you are trying to do both in one operation?
I did it in two separate chamfers.

Kuzma30, 2022-04-16:
Update.
I previously use Face for chamfer top box and it not work. Now I try select Edge. For internal pocket it works. Then I select all external Edges. It show error. Then start some magics. I delete external edges from list and repeat selecting of some ext. edge. Now it works.

[image]

bambukouk, 2022-04-16:
I was going to share this:
[image]