realthunder/FreeCAD #359: The problem of freecad crashing using sublink.

hamza00789, 2022-04-17:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)
 

Freecad crashes when I use the sublink command in the lattice workbench. As you can see in the video free cad crashes when I try to sublink a face. 

https://user-images.githubusercontent.com/92737149/163720003-3d30c86d-25b6-44a0-ad9f-70afb3744879.mp4

closed 2022-04-18 by hamza00789

realthunder, 2022-04-18:
The above commit fixes the crash. But the sub link in lattice probably won't work in this branch. @DeepSOIC any reason to use zero based index for element reference? The topo naming in my branch only works with one based index. The current implementation will omit the 0 index of any element. I can modify the code to keep the zero, but this `wrong` index may actually cause topo naming to incorrectly change the reference in case of model change.

DeepSOIC, 2022-04-18:
It's a hack. Face0 actually refers to Face6, i.e. the last face. Face5 is Face-1. This hack is used in Lattice2 in a few places (i think), i suppose i'll have to fix it. Maybe it's time to blow the dust off of Lattice3, so that i can fix it without worrying of backwards compatibility.

hamza00789, 2022-04-18:
> It's a hack. Face0 actually refers to Face6, i.e. the last face. Face5 is Face-1. This hack is used in Lattice2 in a few places (i think), i suppose i'll have to fix it. Maybe it's time to blow the dust off of Lattice3, so that i can fix it without worrying of backwards compatibility. 

If you can fix it, please fix it.

commit 9dc1a29d (2022-04-18): App: fix PropertyLinkBase::isSame()

Fixes #359
commit 58b7f7ff (2022-04-26): App: fix PropertyLinkBase::isSame()

Fixes #359