realthunder/FreeCAD #361: tree view overlay prevents double clicking on constraint in sketcher

kevenwyld, 2022-04-18:
See attached video.  When using transparent overlay one can not double click on a constraint if it is underneath the transparent area.  If the constraint is moved out from under the overlay it can be double clicked to open the dialog just fine.

https://user-images.githubusercontent.com/11936535/163752899-ab30162a-3ddd-473f-980b-ac7c5ac124f7.mp4

As always thanks for all your hard work!

```
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

realthunder, 2022-04-18:
I can't reproduce the problem. Can you please send me you `link.user.cfg` file (or `Tools -> Edit parameters -> Export`).

kevenwyld, 2022-04-18:
Sure, I used the parameters export feature, here's the file it produced: 
[current_parameters.FCParam.zip](https://github.com/realthunder/FreeCAD/files/8503655/current_parameters.FCParam.zip)

(the file is actually zipped)

Thanks for the help

realthunder, 2022-04-18:
The problem is fixed by the commit above. As a work around, you can enable the following option.

[image]

commit 4a421a4d (2022-04-18): Gui: fixes mouse double click pass through in transparent overlay

Fixes #361
closed 2022-04-18 by realthunder