realthunder/FreeCAD #368: 2 issues with the Iconset feature

HaasNL2, 2022-04-19:
I was making some Icons for Freecad, and stumbled on 2 issues for now.
Thought I report them before I forget. I think the image is self explanatory.
The didn't work just did nothing. Probably issue with the drop-out toolbar thingy.
[image]
`OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
`

luzpaz, 2022-04-19:
Do you have a branch that we can look at the changes you've made?

HaasNL2, 2022-04-19:
[MyIcons.txt](https://github.com/realthunder/FreeCAD/files/8515507/MyIcons.txt)
[image]
[image]
Added the Icon txt file and the svg of the one that didn't work. 
It's not a branch just an Icon set.

realthunder, 2022-04-20:
They are not the same icons. You can filter using keyword body and see there are two icons. Mouse over the icon will show its context, i.e. under what context the icon is used. You can see the icon `PartDesign_Body_Tree` is the one used by feature `PartDesign::Body`. You can find out the feature type through Python command `App.ActiveDocument.Body.TypeId`. For Python extension object, for example my Assembly3 container, you can get the type as `App.ActiveDocument.Assembly.Proxy.__class__()`.

[image]

HaasNL2, 2022-04-20:
Ha ok, so it kinda of a puzzle to get this right.
I noticed that if there is no SVG file in the icon folder it just skips the custom icon, wouldn't it then not be smart to just make a master list with almost all Icons?
I have now something like this:
`PartDesign_Pad, iconset:MyIcons/PartDesign_Pad.svg
PartDesign_Revolution, iconset:MyIcons/PartDesign_Revolution.svg
PartDesign_AdditiveLoft, iconset:MyIcons/PartDesign_AdditiveLoft.svg
PartDesign_AdditivePipe, iconset:MyIcons/PartDesign_AdditivePipe.svg
PartDesign_AdditiveHelix, iconset:MyIcons/PartDesign_AdditiveHelix.svg
PartDesign_AdditiveBox, iconset:MyIcons/PartDesign_AdditiveBox.svg
PartDesign_Body, iconset:MyIcons/PartDesign_Body.svg
PartDesign_Pocket, iconset:MyIcons/PartDesign_Pocket.svg
PartDesign_Hole, iconset:MyIcons/PartDesign_Hole.svg
PartDesign_Groove, iconset:MyIcons/PartDesign_Groove.svg
PartDesign_SubtractiveLoft, iconset:MyIcons/PartDesign_SubtractiveLoft.svg
PartDesign_SubtractivePipe, iconset:MyIcons/PartDesign_SubtractivePipe.svg
PartDesign_SubtractiveHelix, iconset:MyIcons/PartDesign_SubtractiveHelix.svg
PartDesign_Pad_Tree.svg, iconset:MyIcons/PartDesign_Pad.svg
PartDesign_Revolution_Tree.svg, iconset:MyIcons/PartDesign_Revolution.svg
PartDesign_AdditiveLoft_Tree.svg, iconset:MyIcons/PartDesign_AdditiveLoft.svg
PartDesign_AdditivePipe_Tree.svg, iconset:MyIcons/PartDesign_AdditivePipe.svg
PartDesign_AdditiveHelix_Tree.svg, iconset:MyIcons/PartDesign_AdditiveHelix.svg
PartDesign_AdditiveBox_Tree.svg, iconset:MyIcons/PartDesign_AdditiveBox.svg
PartDesign_Body_Tree.svg, iconset:MyIcons/PartDesign_Body.svg
PartDesign_Pocket_Tree.svg, iconset:MyIcons/PartDesign_Pocket.svg
PartDesign_Hole_Tree.svg, iconset:MyIcons/PartDesign_Hole.svg
PartDesign_Groove_Tree.svg, iconset:MyIcons/PartDesign_Groove.svg
PartDesign_SubtractiveLoft_Tree.svg, iconset:MyIcons/PartDesign_SubtractiveLoft.svg
PartDesign_SubtractivePipe_Tree.svg, iconset:MyIcons/PartDesign_SubtractivePipe.svg
PartDesign_SubtractiveHelix_Tree.svg, iconset:MyIcons/PartDesign_SubtractiveHelix.svg
PartDesign_Pad.svg, iconset:MyIcons/PartDesign_Pad.svg
PartDesign_Revolution.svg, iconset:MyIcons/PartDesign_Revolution.svg
PartDesign_AdditiveLoft.svg, iconset:MyIcons/PartDesign_AdditiveLoft.svg
PartDesign_AdditivePipe.svg, iconset:MyIcons/PartDesign_AdditivePipe.svg
PartDesign_AdditiveHelix.svg, iconset:MyIcons/PartDesign_AdditiveHelix.svg
PartDesign_AdditiveBox.svg, iconset:MyIcons/PartDesign_AdditiveBox.svg
PartDesign_Body.svg, iconset:MyIcons/PartDesign_Body.svg
PartDesign_Pocket.svg, iconset:MyIcons/PartDesign_Pocket.svg
PartDesign_Hole.svg, iconset:MyIcons/PartDesign_Hole.svg
PartDesign_Groove.svg, iconset:MyIcons/PartDesign_Groove.svg
PartDesign_SubtractiveLoft.svg, iconset:MyIcons/PartDesign_SubtractiveLoft.svg
PartDesign_SubtractivePipe.svg, iconset:MyIcons/PartDesign_SubtractivePipe.svg
PartDesign_SubtractiveHelix.svg, iconset:MyIcons/PartDesign_SubtractiveHelix.svg`

HaasNL2, 2022-04-20:
Still can't get PartDesign_AdditiveBox or PartDesign_SubtractiveBox or PartDesign_CompPrimitiveAdditive to work.

HaasNL2, 2022-04-20:
Update [image]

realthunder, 2022-04-21:
Added a few commits to provide a more consistent icon cache key. It is not really possible to provide a master icon list like you suggested. FreeCAD has a modular design. The core provides some facility for other modules to store and retrieve icons. So there is no way to know beforehand how many icons there are, and where those icons are used for. The core provides a way for other module to supply some usage context, but it is up to the module whether to supply the context or not. Example context are, object type name, python object class name, command name, resource path. You can type any of those possible context in the icon browser filter to search for possible match.

HaasNL2, 2022-04-21:
I have added my first release of my Icon pack.
[MisterMaker_icons.zip](https://github.com/realthunder/FreeCAD/files/8529447/MisterMaker_icons.zip)

HaasNL2, 2022-04-23:
Don\t want to open a new issue for this and it is just a question but is it possible to change the way the WB icons are positioned, I'd like to use rectangle icons not square: 
[image]

HaasNL2, 2022-04-24:
This is awesome, I'm gonna make it look so good people are never going to believe it is  Freecad :)
Only risk is that people are going to sell it. :)

closed 2022-04-24 by HaasNL2