realthunder/FreeCAD #377: File and Edit menu items are not clickable.

asoee, 2022-04-24:
The File and Edit menu items (and some of View) in the main menu are not clickable.
It seems like there is a transparent overlay, that blocks mouse interactions.
If I click on Tools, and move the mouse left, they do open as they should.

[image]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.407.28576 (Git)
Build type: Release
Branch: LinkStage3
Hash: 599f7663af69d3dbd77063e834b12f26da513b2d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Denmark (en_DK)

realthunder, 2022-04-24:
Sounds similar to #335. But I can't seem to reproduce it.

asoee, 2022-05-02:
Still present in 2022.04.30 version (both daily and stable)
I managed to grab this, using the windows inspect tool:

[image]

If that is any help...

realthunder, 2022-05-03:
Hmm... a QTabBar without an empty name. Maybe its from some external workbench you've installed? Can you please send me you user configuration `Tools -> Edit parameters -> Export`? You can try reset your user configuration and see if it helps, `Tools -> Edit parameters -> Reset`. You restore the configuration using `Import`.

asoee, 2022-05-06:
Resetting the parameters solved it.
So i guess this is not linkstage3 related.

Even when re-importing the parameters, i cannot reproduce the problem, but here they are, anyways:
[QTabBar.zip](https://github.com/realthunder/FreeCAD/files/8640263/QTabBar.zip)

closed 2022-05-06 by asoee