realthunder/FreeCAD #380: UTF encoding

ljungis, 2022-04-26:
Hello,
I am running Freecad link branch vers. 2022.426 on a Linux mint 20.3, Cinnamon, 64-bit.
When I try to extrude a sketch up surface I get the following message. 

("(unicode error) 'utf-8' codec can't decode byte 0xf6 in position 5: invalid start byte", ('<string>', 1, 77, None))

What can be done?

Thanks for any help.

realthunder, 2022-04-26:
Can you please post a file in the state just before the extrusion? Also, it would be better to have a screencast showing your steps?

realthunder, 2022-04-26:
The error is most likely related to locale setting. Which language are you using? Maybe post your user configuration here as well, `Tools -> Edit parameters -> Export`.

ljungis, 2022-04-26:
[Skärmbild från 2022-04-26 16-29-26.png.zip](https://github.com/realthunder/FreeCAD/files/8564214/Skarmbild.fran.2022-04-26.16-29-26.png.zip)
[Tools.FCParam.zip](https://github.com/realthunder/FreeCAD/files/8564215/Tools.FCParam.zip)
[Extrude_test.zip](https://github.com/realthunder/FreeCAD/files/8564221/Extrude_test.zip)

I am using swedish.

ljungis, 2022-04-26:
I recently noticed that if you save the file before performing the extrusion, close the file and open it again, the extrusion will be performed correctly.

realthunder, 2022-04-27:
Can you please paste the FreeCAD About information here?

ljungis, 2022-04-27:
OS: Ubuntu Core 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.426.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.26-edge)
Hash: e2d286fb97c9a81c741e6a0d102f7cae2065f6f7
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: Swedish/Sweden (sv_SE)

I have made a few further attempts and found that if I, anytime before the extrusion, save and close the current part, the extrusion can be carried out without error when the part is reopened

realthunder, 2022-04-27:
It is because unsaved file is named as `Unamed`, or the equivalent word in your current language settings, which happens to contain some non ascii alphabet. It could be easily fixed if I can find where the error is emitted, but I have a hard time finding it. Anyway, I have made a large scale change to try to fix the character encoding everywhere. Hope it can solve the problem. You try it a bit later when the snap build is finished.

ljungis, 2022-04-27:
I have now updated to version 2022.427 and the problem is gone.
Thank you very much!!

closed 2022-04-30 by realthunder

ljungis, 2022-10-11:
OS: Ubuntu Core 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.426.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.26-edge)
Hash: e2d286fb97c9a81c741e6a0d102f7cae2065f6f7
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: Swedish/Sweden (sv_SE)

 

 

I have made a few further attempts and found that if I, anytime before the extrusion, save and close the current part, the extrusion can be carried out without error when the part is reopened.

 
 

Sent: Wednesday, April 27, 2022 at 3:21 AM