realthunder/FreeCAD #390: Can't open or create a new project

m-sundman, 2022-05-03:
`F[reeCAD-asm3-Stable-Conda-Py3-Qt5-20220430-glibc2.12-x86_64.AppImage` doesn't work for me at all. Trying to open an existing project, or even create a new project, causes the whole FreeCAD window to just freeze. Nothing is written to stdout when this happen, but when I start it the following gets written:
```
/tmp/.mount_FreeCAwxEJb8
FreeCAD 0.19, Libs: 2022.430R28641 (Git)
libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
4.346073 <asm3.main> init_gui.py(14): no solver backend found
```
I'm using integrated intel graphics.

m-sundman, 2022-05-03:
The same thing happens with `FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220430-glibc2.12-x86_64.AppImage`

realthunder, 2022-05-04:
Looks similar to kiwix/kiwix-desktop#393. Please check out the solutions posted there. Either try specify a different graphics driver, or preload your system libstdc++ and see if it works.

m-sundman, 2022-05-04:
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6` didn't work.
`MESA_LOADER_DRIVER_OVERRIDE=i915` (which matches my iGPU) causes it to segfault immediately when creating/opening a project.
`MESA_LOADER_DRIVER_OVERRIDE=i965` seems to work. Creating a new project works. Opening an existing project works.

closed 2025-05-09 by m-sundman