realthunder/FreeCAD #391: External geometry fails

m-sundman, 2022-05-05:
Projecting curves using the "external geometry" tool fails in 2022.430 (28641).

Here projecting the yellow curve causes the line to go down to y=0, instead of being at the correct y, like the projected straight line:
[image]
Both lines should be at the same height (where the longer line is).
What's worse, editing a sketch saved in a previous version will also cause pre-existing projected external geometry to jump this way, completely messing up the design.

realthunder, 2022-05-05:
Can you please post the sample file here, and also the `About` information.

m-sundman, 2022-05-05:
Here's the project above:
[ref-error.FCStd.zip](https://github.com/realthunder/FreeCAD/files/8628484/ref-error.FCStd.zip)

By "About information", do you mean the version (2022.430) and build number (28641) I included above, or what?

realthunder, 2022-05-05:
I mean the `About` dialog of the application, `Help -> About FreeCAD -> Copy to Clipboard`.

m-sundman, 2022-05-05:
Ah, sorry, here you go:

OS: Ubuntu 22.04 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.430.28641 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: a77045488092c24ece25d63c91f362cbd2181d36
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

closed 2022-05-05 by realthunder

m-sundman, 2022-05-05:
Is the "daily" version rebuilt daily? I.e., if I download it tomorrow, will it include this fix?

realthunder, 2022-05-05:
You can try the [snap](https://snapcraft.io/freecad-realthunder) build tomorrow. It's auto built by github workflow, and hence will be updated much more frequently. The snap daemon also auto refresh your installation in the background. However, due to its default sandbox feature, you may find a bit cumbersome to use. Anyway, this is a serious problem. I'll update AppImage soon.

m-sundman, 2022-05-05:
Thanks for fixing it so quickly. Snaps are just awful to use, so I'll grudgingly but eagerly wait for the AppImage. :)

commit 3b51a1ae (2022-05-05): Sketcher: fix external geometry arc projection

Fixes #391
@m-sundman
commit fd916016 (2022-05-10): Sketcher: fix external geometry arc projection

Fixes #391