realthunder/FreeCAD #392: Issue with Path Workbench tool library on snap version

ngocminhdao88, 2022-05-05:
Hi,

FreeCAD stops to repond when i agree to setup the Toolbit working dir in Path Workbench. I have to kill it to restart FreeCAD again.

[image]

By saying no i got to this page. But clicking on first and third button does nothing. On second button i got this error (see below)

[image]

>  Traceback (most recent call last):
18:12:48    File "/snap/freecad-realthunder/27/usr/Mod/Path/PathScripts/PathToolBitLibraryGui.py", line 623, in libraryNew
18:12:48      with open(path, 'w') as fp:
18:12:48  PermissionError: [Errno 13] Permission denied: '.fctl'
18:17:03  Traceback (most recent call last):
18:17:03    File "/snap/freecad-realthunder/27/usr/Mod/Path/PathScripts/PathToolBitLibraryGui.py", line 650, in libraryOk
18:17:03      self.librarySave()
18:17:03    File "/snap/freecad-realthunder/27/usr/Mod/Path/PathScripts/PathToolBitLibraryGui.py", line 646, in librarySave
18:17:03      with open(self.path, 'w') as fp:
18:17:03  OSError: [Errno 30] Read-only file system: '/snap/freecad-realthunder/27/usr/Mod/Path/Tools/Library/Default.fctl'
18:17:07  Traceback (most recent call last):
18:17:07    File "/snap/freecad-realthunder/27/usr/Mod/Path/PathScripts/PathToolBitLibraryGui.py", line 650, in libraryOk
18:17:07      self.librarySave()
18:17:07    File "/snap/freecad-realthunder/27/usr/Mod/Path/PathScripts/PathToolBitLibraryGui.py", line 646, in librarySave
18:17:07      with open(self.path, 'w') as fp:
18:17:07  OSError: [Errno 30] Read-only file system: '/snap/freecad-realthunder/27/usr/Mod/Path/Tools/Library/Default.fctl'
18:17:11  Traceback (most recent call last):
18:17:11    File "/snap/freecad-realthunder/27/usr/Mod/Path/PathScripts/PathToolBitLibraryGui.py", line 650, in libraryOk
18:17:11      self.librarySave()
18:17:11    File "/snap/freecad-realthunder/27/usr/Mod/Path/PathScripts/PathToolBitLibraryGui.py", line 646, in librarySave
18:17:11      with open(self.path, 'w') as fp:
18:17:11  OSError: [Errno 30] Read-only file system: '/snap/freecad-realthunder/27/usr/Mod/Path/Tools/Library/Default.fctl'

My FreeCAD version:

> OS: Ubuntu Core 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 2022.501.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.30-edge)
Hash: 36be73c9b78b77e544f9522c220a9b929715ce90
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)

realthunder, 2022-05-06:
Please try set the working directory, as it seems the default on is the same directory of the source code which is readonly in case of snap or AppImage.

luzpaz, 2022-07-06:
@ngocminhdao88 can you test on latest edge snap ?

ngocminhdao88, 2022-07-16:
> @ngocminhdao88 <https://github.com/ngocminhdao88> ping

Hi lupaz,

I dont have access to my PC right now so i can not test. I will try when i
am back to work.

Best regards,
NMD

discip, 2022-10-05:
@ngocminhdao88
Is this still an issue?
If not, please close this one. 😊

closed 2022-10-06 by ngocminhdao88