realthunder/FreeCAD #393: Segfault when creating image plane

m-sundman, 2022-05-05:
Not sure which particular program/library is at fault here.

Here's the top of the stack on FreeCAD 2022.407 Stable:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f97a62a6520]
#1  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x27857f) [0x7f979aed057f]
#2  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x2bd369) [0x7f979af15369]
#3  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x2c19e2) [0x7f979af199e2]
#4  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x52424e) [0x7f979b17c24e]
#5  0x7f97a8968d2d in NaviCubeShared::drawNaviCube(SoCamera*, bool, int, bool) from /tmp/.mount_FreeCAdX74YO/usr/bin/../lib/libFreeCADGui.so+0x13cd
#6  0x7f97a8969146 in NaviCubeImplementation::drawNaviCube(bool) from /tmp/.mount_FreeCAdX74YO/usr/bin/../lib/libFreeCADGui.so+0x46
#7  0x7f97a8933ef5 in Gui::View3DInventorViewer::renderScene() from /tmp/.mount_FreeCAdX74YO/usr/bin/../lib/libFreeCADGui.so+0x3f5
#8  0x7f97a88e6c1d in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAdX74YO/usr/bin/../lib/libFreeCADGui.so+0x31d
#9  0x7f97a88ecea5 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAdX74YO/usr/bin/../lib/libFreeCADGui.so+0x45
#10  0x7f97a706895d in QWidget::event(QEvent*) from /tmp/.mount_FreeCAdX74YO/usr/bin/../lib/libQt5Widgets.so.5+0x5a3
...
```
OS: Ubuntu 22.04 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.407.28576 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 599f7663af69d3dbd77063e834b12f26da513b2d
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

Unsurprisingly it also happens with other versions.

Here's the top of the stack running FreeCAD 2022.430 Stable:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f28289f4520]
#1  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x27857f) [0x7f281d6db57f]
#2  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x2bd369) [0x7f281d720369]
#3  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x2c19e2) [0x7f281d7249e2]
#4  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x52424e) [0x7f281d98724e]
#5  0x7f282b0beaed in NaviCubeShared::drawNaviCube(SoCamera*, bool, int, bool) from /tmp/.mount_FreeCAnaPeXN/usr/bin/../lib/libFreeCADGui.so+0x13cd
#6  0x7f282b0bef06 in NaviCubeImplementation::drawNaviCube(bool) from /tmp/.mount_FreeCAnaPeXN/usr/bin/../lib/libFreeCADGui.so+0x46
#7  0x7f282b089cb5 in Gui::View3DInventorViewer::renderScene() from /tmp/.mount_FreeCAnaPeXN/usr/bin/../lib/libFreeCADGui.so+0x3f5
#8  0x7f282b03c9ad in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAnaPeXN/usr/bin/../lib/libFreeCADGui.so+0x31d
#9  0x7f282b042c65 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAnaPeXN/usr/bin/../lib/libFreeCADGui.so+0x45
#10  0x7f28297b695d in QWidget::event(QEvent*) from /tmp/.mount_FreeCAnaPeXN/usr/bin/../lib/libQt5Widgets.so.5+0x5a3
...
```
OS: Ubuntu 22.04 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.430.28641 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: a77045488092c24ece25d63c91f362cbd2181d36
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

Shkolik, 2022-07-15:
Could not reproduce on:

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)

m-sundman, 2025-05-09:
Seems fixed

closed 2025-05-09 by m-sundman