realthunder/FreeCAD #395: Part design features moved out of object and cannot be moved back in

elcojacobs, 2022-05-06:
I somehow ended up with a document that has part design features that are not inside a body.
The migrate function says it cannot find objects without a body. 
If I try to move them into a body, it complains about dependencies. If I try to copy them, Freecad crashes.

Can I somehow make changes to the XML to fix this?

The file contains some information that cannot be publicly shared yet. Can I send it to you in a PM or give you access to the private GitHub repo?

elcojacobs, 2022-05-06:
I have invited you to the repo. The latest commit breaks the document. Unfortunately, that single commit has many hours of work so I would prefer fixing the links instead of reverting the commit.

realthunder, 2022-05-06:
I'll check this out soon. Can you recall the steps you've done that led to
the current state?

elcojacobs, 2022-05-06:
I have opened the document with regular Freecad weekly, but then noticed that multiple solids were not supported so I closed it again without saving and went back to your branch.

realthunder, 2022-05-08:
I have pushed the fixed document to your repo. Please check. Next time, try to remember the steps.

PS. if you are using git to host your project, maybe you can try `Save as directory` option to save your document into a separate directory without compression. You can then git control that directory. Sub project can exist as sub directory. Be aware though, adding new feature may create new files in the directory. You'll have to manually add those files to git repo. And remember to save and commit frequently for easy going back.

I'll eventually create some workbench to provide integrate support of version control some time in the future.

elcojacobs, 2022-05-08:
Thank you so much! How did you fix it?

The 'save as directory' option sounds better for version control indeed. I'll switch to that.
I really like your version of Freecad, keep up the great work!

closed 2022-05-08 by elcojacobs

elcojacobs, 2022-05-09:
The order of features seems wrong in the fixed commit, which makes the doc slow to recompute.
In trying to address that, I found a way to reproduce.

The very first thing in the body is Sketch 129, which is just an empty sketch without any dependencies.

[image]

Deleting sketch 129 pushes all other features out of the body :open_mouth:

realthunder, 2022-05-10:
I've made another push to include some missed features. The reason of those features being moved out is not because deleting sketch129. It is that because of deletion cause `Body.Group` property being modified and triggers link scope checking. The body (as well as `App::Part`) demands all features only references features inside itself. You should also be careful not to reference object inside the body container from outside the container (e.g. you've references features inside the body to create those curves that are outside of the body). These are called cross coordinate references, because each body contain defines its own coordinate system, and those outside the body is in the global coordinate system. There are special objects that allows for cross coordinate references, such as `SubShapeBinder`. There is also an option to disable cross coordinate check in `Preferences -> General -> Objects -> Allow cross coordinate links`.

Here is how I fixed your document. First, enable the above mentioned `Allow cross coordinate links` option. Then scroll down the tree view and find object `Import (back_core_xy)`, and select this and allow objects below. Then run the following code

```python
# Get selected objects
group = Gui.Selection.getSelection()
# obtain all objects that references any objects in the selection
ins = set([o for oo in group for o in [oo] + oo.InListRecursive if not o.getParentGeoFeatureGroup() and (o.isDerivedFrom('PartDesign::Feature') or o.isDerivedFrom('Part::Part2DObject') or  o.isDerivedFrom('Part::SubShapeBinder') or o.isDerivedFrom('Part::Datum'))])
# Obtain all dependencies of the above objects and sort them in dependency order
deps = [o for o in App.getDependentObjects(list(ins), 1) if o not in App.ActiveDocument.Body.Group and not o.getParentGeoFeatureGroup() and (o.isDerivedFrom('PartDesign::Feature') or o.isDerivedFrom('Part::Part2DObject') or  o.isDerivedFrom('Part::SubShapeBinder') or o.isDerivedFrom('Part::Datum'))]
# Add those objects to body
App.ActiveDocument.Body.Group += deps
```

elcojacobs, 2022-05-10:
I did notice the warnings about cross-links, but I don't know how to do it properly.

The mixed curves from the curves workbench cannot exist in the PartDesign body. I would rather have them inside, but this doesn't seem to be supported. 
The 2 sketches that they use as their input are dependent on many things inside the PartDesign body, so they are inside the body. The result of the mixed curve is outside and wrapped as a SubShapeBinder binder to get them back in.

This in and out of feature required for the curves seems to be able to have 'SubShapeBinder' for back in, but no binder for part design to part WB?

realthunder, 2022-05-11:
The binder can be used on any feature with a Shape, whether it's from PartDesign or Part or some other workbench. You can create a binder using the global structure toolbar as shown below,

[image]

You can also create external feature inside body without using binder. You need to make sure the body is active, and new external feature only depends on features inside the body. For example, in your case, after activate the body, select two sketches inside the body, and click `Mixed curve`, a dialog will prompt asking if you want to wrap the new feature inside the body as shown below.

[image]

If you didn't see the dialog, maybe you've rejected it before and made it remember the choice. You can revert the setting using the following preference option,

[image]

elcojacobs, 2022-05-11:
Thank you so much for the insights. I was able to clean up my document and get all the features in scope. I had indeed disabled auto wrap, but have changed this setting to always enable it now.