realthunder/FreeCAD #397: could not open files or folders from mounted disk

mwlandolsi, 2022-05-08:
When navigating to a mounted disc or device, then it is displayed empty
[image]
[image]

OS: Ubuntu Core 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.505.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.05.05-2-edge)
Hash: 32adb43c3aa04846a3fb2b019f8efdfe3a06545f
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)

mwlandolsi, 2022-05-08:
It is link daily installed from snap-store. This issue does not exist in .appimage version.

luzpaz, 2022-05-08:
> It is link daily installed from snap-store. This issue does not exist in .appimage version.

CC @ppd
Hi! Any suggestions regarding accessing mounted disks with Snap ?

ppd, 2022-05-08:
I'd check whether the removable storage devices permission has been granted in the software store thingy:

[image]

Or by terminal:

```shell
snap connect freecad-realthunder:removable-media
```
`

luzpaz, 2022-05-10:
@mwlandolsi ping

mwlandolsi, 2022-05-10:
Hello,
sorry to answer later, but it is not solved. 
Here is what I got after applying the command (terminal)

[image]

I do not know if this could help : 
Notice that I have set the auto-mount option on.

[image]

ppd, 2022-05-12:
@mwlandolsi 
Could you please try the following:

```shell
# install snappy-debug
$ sudo snap install snappy-debug

# run snappy-debug in one terminal
$ sudo snappy-debug

# enter freecad-realthunder in a second terminal
$ snap run --shell freecad-realthunder

# in the second shell, navigate to your external disk
$ cd /home/xxx/Documents/discs/MyDocExt4

# execute some file accessing command, e.g. ls
$ ls -al
```

You should get some errors from `ls -al`, and, more importantly, you should receive some AppArmor errors from `snappy-debug` in the second terminal. Those errors can probably tell why access is being denied. 
Maybe it's about the UID & GID of the mount point within your home directory.

mwlandolsi, 2022-05-12:
Hello Maximilian
I think you mean  the **first** terminal here :
> (...), you should receive some AppArmor errors from `snappy-debug` in the **second** terminal.(....)

Here what I got :

```
= AppArmor =
Time: May 12 10:20:43
Log: apparmor="DENIED" operation="open" profile="snap.freecad-realthunder.freecad-realthunder" name="/home/xxx/.bash_history" pid=19404 comm="FreeCADLink" requested_mask="r" denied_mask="r" fsuid=1000 ouid=1000
File: /home/xxx/.bash_history (read)
Suggestions:
* adjust program to read necessary files from $SNAP, $SNAP_DATA, $SNAP_COMMON, $SNAP_USER_DATA or $SNAP_USER_COMMON
* add 'personal-files (see https://forum.snapcraft.io/t/the-personal-files-interface for acceptance criteria)' to 'plugs'

= AppArmor =
Time: May 12 10:20:50
Log: apparmor="DENIED" operation="open" profile="snap.freecad-realthunder.freecad-realthunder" name="/home/xxx/Documents/discs/MyDocExt4/" pid=19404 comm="QFileInfoGather" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
File: /home/xxx/Documents/discs/MyDocExt4/ (read)
Suggestion:
* add 'home' to 'plugs'
```

Me, and many other users,  are not good in Linux CLI ;) . So please if there is any suggestion, it will be helpful for the community.
Many thanks

ppd, 2022-05-12:
@mwlandolsi 

from the denial, we can see:

>  fsuid=1000 ouid=0

This means that the mount point `/home/xxx/Documents/discs/MyDocExt4/` is owned by root. This is not allowed in the AppArmor profile:

https://github.com/snapcore/snapd/blob/bb08d2137f443c2169a095782fc0cfb33fb27d51/interfaces/builtin/home.go#L59

Therefore, I suggest either changing the owner of said directory & files (if that makes sense), or to mount it in `/mnt` or `/media/<user>` (or `/run/media/<user>`), where snapd is more tolerant wrt file ownership:

https://github.com/snapcore/snapd/blob/bb08d2137f443c2169a095782fc0cfb33fb27d51/interfaces/builtin/removable_media.go

closed 2022-05-23 by mwlandolsi