realthunder/FreeCAD #398: Graphviz's path not found

gharkanyi, 2022-05-08:
Hello!

Snap:
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.505.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.05.05-2-edge)
Hash: 32adb43c3aa04846a3fb2b019f8efdfe3a06545f
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)

It's not a blocker, mainstream problem, but the `Tools -> Dependency graph...` does not work, namely the `Graphviz`'s path is not found at all, although it is specified correctly in the parameters:
[image]
Anyway, the `Graphviz` works from command line, e.g. `$ echo 'digraph { a -> b }' | dot -Tsvg > /tmp/output.svg` is OK and:
[image]

The FC's response:
[image]

Any idea?
Thank you in advance!

Bests
Gábor

realthunder, 2022-05-10:
@ppd Any idea on how to launch external program from a confined snap app?

ppd, 2022-05-10:
@realthunder not really possible aside from xdg-open. That's part of the security model. It's better to ship such dependencies in the snap itself.

ppd, 2022-05-10:
Hm. Graphviz is already in stage-packages. Dot is located in `$SNAP/usr/bin/dot` which expands to `/snap/freecad-realthunder/current/usr/bin/dot`. We could set up a symlink in the mount namespace, so FreeCAD finds `$SNAP/usr/bin/dot` at `/usr/bin/dot`.

ppd, 2022-05-10:
@realthunder Some graphviz fixes are here: https://github.com/ppd/freecad-ppd/commit/3735d73bad74c55b11181336b2f36457d3bb0d20

Let's see if that works out, and then we can integrate it into freecad-realthunder.

luzpaz, 2022-06-15:
FreeCAD master Snap has been fixed in https://github.com/FreeCAD/FreeCAD-snap/commit/3b43c39b47e5311709b329f47e945f87733bd5b9

luzpaz, 2022-07-06:
@gharkanyi please try again on latest edge

gharkanyi, 2022-07-10:
Hi Guys!
Sorry for the somewhat late answer ...

Thank you very much, it WORKS now!
Thanks for yours works on this great project, go on in this way!

All the best!
Gábor

closed 2022-07-10 by gharkanyi