realthunder/FreeCAD #402: Improvements on the Icon themes and some minor issues with Lattice2 icons.

HaasNL2, 2022-05-11:
I really like the icon theme support I now have 5 partially themes for part design and sketcher. And I am planning to finish sketcher too and maybe more if people want.
But after doing all that work I had some improvements requests. 

1.Make it possible to use more than one Icon theme. Because you might want to have an icon theme for PartDesign and another for Sketcher and you maybe found a great one for Path. So you need to be able to combine it. Otherwise you eventually end up with to many combination theme's. I would suggest adding a way to select them in order so the last one will overwrite the second last one and the second last one will overwrite the third etc. , so if you have doubles in other lists it will just takes the last it sees.
I noticed the list already does this if you have multiple instances it just takes the last one. 

2. Add a "open Icon theme directory" to the icon theme setting, because users have issues finding the correct folder and that is not strange since it is a hidden folder by default. Only superusers will find it easily since they already been there before.
 
3. Lattice2 workbench resource paths don't work, I asked the @[DeepSOIC](https://github.com/DeepSOIC) if you could fix the resource paths but he was unable. Maybe someone can have a look? I checked some other workbenches and those had no issues. https://github.com/DeepSOIC/Lattice2/issues/65#issuecomment-1120297244

If someone asks himself why we need this well I have been working hard to make Icon theme's and it is awesome: https://mistermaker.gumroad.com/l/Linkstage_Icons

realthunder, 2022-05-13:
The first column of the icon list is an arbitrary user code selected name for the icon. It can be anything to uniquely identify the icon. I tried to replace some icon for Lattice, and it seem to work just fine. Did you encounter any problem?

Regarding the composite icon theme, maybe I can add some `import` statement to include other icon sets. I'll think about it.

HaasNL2, 2022-05-13:
@realthunder after your comment I rechecked the lattice Icon, and it seems that I made a typo..... Facepalm.

discip, 2022-10-05:
@MisterMakerNL
> after your comment I rechecked the lattice Icon, and it seems that I made a typo..... Facepalm.

Than please close this issue. 😊

closed 2022-10-06 by MisterMakerNL