realthunder/FreeCAD #405: Installing daily releases in Windows for Beginners

mszlazak, 2022-05-14:
Hello, can you provide detailed instructions on how to get a latest daily release running on Windows 10?
This would be for a beginner. Basically someone that does not know to much technical stuff about computers and software. What Python 3 would be needed (i have the latest) and where is it located? What other things do i need preinstalled? Here is a Youtube that attemps to teach beginners how to do this but the method fails ob your latest release (2022.04.30) or daily. 

https://youtu.be/p_ZEry2wTfg

bambukouk, 2022-05-14:
in what way does it fail?

mszlazak, 2022-05-14:
Well i installed Python 3 latest (3.10 i believe). Set up some folders to install solvespace as Realthunder recommended. Set up another folder in the location Realthunder recommended and unzipped his software into it. In that folder i clicked on a file ending in .bin but nothing happened. The video i linked to clicks on a file with a different extension. That is badically it.

bambukouk, 2022-05-15:
1.download FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220430.7z to a folder of your choice
2.extract it
3.open bin folder
4. find FreeCADLink.exe
5. either run it directly or create shortcut

You don't need to install Python
You don't need to create folder to install solvespace
It's not `clicked on a file ending in .bin` it's a bin folder inside of which you will find .exe
Don't know where you got this instructions  `as Realthunder recommended` ?

I normally use Linux, but just installed latest daily in my Win10 machine (as per above) and it is working perfectly OK.

mszlazak, 2022-05-15:
Thank you! Your instructions gets FreeCAD up and running. Hopefully it all works.
I made a mistake and said I clicked on the .bin file but i meant to write .bat file ... no matter now.
Next are these instructions that realthunder has posted which made me set up folders in different locations and separate out solvespace. 

https://github.com/realthunder/FreeCAD_assembly3

bambukouk, 2022-05-15:
> Next are these instructions that realthunder has posted which made me 
> set up folders in different locations and separate out solvespace.

Got it...
Yes, it can be confusing - these instructions are out of date and mostly applicable to the users of master branch trying to install Assembly3 as an addon.

You will still find, that the first time you try to use Assembly3 it will ask you about downloading the solver - just follow the instructions and the process is automatic.

Glad it works for you (and join us on the forum as linked previously :+1: )

If it is all OK, you can (should?) close this issue as a solved one.

closed 2022-05-15 by mszlazak

mszlazak, 2022-10-11:
I cannot even get started. Nothing happens!