realthunder/FreeCAD #407: Value input dialog misses the first keystroke

kaushikbm, 2022-05-16:
Version 2022.430 and previous one. In sketcher some value input dialogs misses the first keystroke. If you type again it works. Somehow it looks that it is not in focus.

[image]

bambukouk, 2022-05-17:
It's most likely you and your hardware (what are you using?)

kaushikbm, 2022-05-17:
> It's most likely you and your hardware (what are you using?)

I'm using an i5 with 16GB RAM and a 4GB NVIDIA graphics card.

realthunder, 2022-05-17:
It is because of a new feature for resolving ambiguous shortcut that have common prefix. The sketcher Radius shortcut is `R`. There is also a global shortcut `R, R` for recent command. So what FreeCAD does when it receives the first `R` is to wait for a while and only trigger shortcut `R` either on timeout or receiving another non `R` key press. Unfortunately that non `R` key is discarded because the input dialog isn't there when the key is pressed. The solution is to either change the Radius shortcut, or shorten the shortcut sequence timeout in `Tools -> Customize -> Keyboard -> Multi-key sequence delay`.

bambukouk, 2022-05-17:
That explains why I haven't noticed - I seldom use radius.... :+1: 
Thank you!
Shouldn't radius shortcut by changed by default rather than leaving to a user to discover this odd fault and remedy it?
I take it, it also applies to master branch rather than uniquely to Link branch?
if yes, should I raise it on the forum?

realthunder, 2022-05-17:
The default shortcut is set by sketcher developer. I've mentioned this new shortcut resolver to them. I even submitted a PR for the resolver. Unfortunately, it is not accepted for 0.20 due to its complexity. Just have to wait.

bambukouk, 2022-05-17:
> ...Just have to wait.
no problem, thank you :+1:  as usual

kaushikbm, 2022-05-17:
> It is because of a new feature for resolving ambiguous shortcut that have common prefix. The sketcher Radius shortcut is `R`. There is also a global shortcut `R, R` for recent command. So what [...]

Thank you so much for clearing the issue.

kaushikbm, 2022-05-17:
> That explains why I haven't noticed - I seldom use radius.... 👍 (and I don't use shortcuts to enter dims) Thank you! Shouldn't radius shortcut by changed by default rather than leaving to a user to discover this odd fault and remedy it? I take it, it also applies to master branch rather than uniquely to Link branch? if yes, should I raise it on the forum?

I'm a shortcut user for CAD programs. Love to keep one hand on keyboard and the other on the mouse. 😃

bambukouk, 2022-05-17:
> I'm a shortcut user for CAD programs

lots of people are, and I understand it - I guess I must be odd (I am a secret mouse abuser...)

kaushikbm, 2022-05-17:
> (I am a secret mouse abuser...)

✌😂

closed 2022-05-18 by kaushikbm