realthunder/FreeCAD #410: Custom Work Plane Selection Not Working

cadmarx, 2022-05-16:


cadmarx, 2022-05-18:
Could anyone please let me know where this post has been moved to? Sorry I am not sure about the rules here.

bambukouk, 2022-05-18:
> Could anyone please let me know where this post has been moved to? Sorry I am not sure about the rules here.
[image]

Clearly you know where it is (since you are posting here :+1: )
Your issue pertains to FreeCAD Linkstage3 branch (and not to Assembly3) and that's where it belongs :+1: 
It is a bit confusing at first...

discip, 2022-10-05:
@cadmarx
Please close this issue as you were directed to the right spot. 😊

closed 2022-10-09 by cadmarx