realthunder/FreeCAD #411: Freecad crashes issue.

hamza00789, 2022-05-17:
Any file that is created in freecad if the file is large and is edited. So in that case, free cad crashes.  

[FreeCAD.zip](https://github.com/realthunder/FreeCAD/files/8708191/FreeCAD.zip)

hamza00789, 2022-05-17:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)
 
The tree length is large.

bambukouk, 2022-05-17:
> The tree length is large

What do you consider `large` ?
can you share the file? for others to test...

hamza00789, 2022-05-17:
> 

I encountered this problem in two separate files. Sorry, unfortunately, I can't provide both files for some reason.

luzpaz, 2022-07-15:
Crash log isn't enough. Can you send the file to realthunder via email ?

closed 2022-07-16 by hamza00789

hamza00789, 2022-07-16:
I don't have the authority to give the file because the file belongs to someone else.

luzpaz, 2022-07-16:
@hamza00789 can you simply the bug so we can reproduce it ?

hamza00789, 2022-07-16:
I don't know exactly what the bug is yet. But I will try to find out and when I know I will discuss this issue.