realthunder/FreeCAD #419: importing dxf from a python script. Help needed.

wsteffe, 2022-05-19:
Hi all,
 using a python script,  I would like to import a bunch of dxf files a and to put the related sketches into separate bodies.

I have already prepared a bunch of dxf files.

Following lines works excluding the commented line which should import one dxf file.
The dxf file import is OK when I am using the graphical interface.

```
import FreeCAD
import importDXF
doc = FreeCAD.newDocument()
doc.saveAs(fname+'.FCStd')

for name in layer_names:
   body=doc.addObject("PartDesign::Body", name)
   #importDXF.insert(fname+"_"+name+".dxf",fname)
   #Here I would like to move the sketches into the added body.

doc.save()
FreeCAD.closeDocument(doc.Name)

```

realthunder, 2022-05-19:
Make sure you've enabled grouping DXF import option as shown below, which will produce one object per layer. You can then use `Draft.makeSketch()` to convert the produced object into sketch. You can then add the sketch into the body.

[image]

```python
import FreeCAD
import importDXF, Draft
doc = FreeCAD.newDocument()
doc.saveAs(fname+'.FCStd')

for name in layer_names:
   body=doc.addObject("PartDesign::Body", name)
   existings = doc.Objects
   importDXF.insert(fname+"_"+name+".dxf",doc.Name)
   newObjs = [o for o in doc.Objects if o not in existings]
   for obj in newObjs:
      sketch = Draft.make_sketch(obj, autoconstraints=True)
      doc.removeObject(obj.Name)
      body.Group += [sketch]
```

wsteffe, 2022-05-19:
Thanks realthunder.  Your receipt worked very well.

My purpose is to import in FreeCAD a layered structure made with klayout.
Here following is my script in the current status. I need to add a few code lines to read the z coordinates zstart,zstop associated with each layer in the stackup. These values will be used to extrude the sketches stored in bodies making solid shapes (or holes).
To use this code you need to have pya module of klayout and you need to build it using the provided build.sh.
In my opinion klayout is simpler than KiCAD. But of course it is subjective.

```
#!/usr/bin/python3
import sys,os
import pya
argv=sys.argv
argl=len(argv)

fname,fext=os.path.splitext(argv[argl-1])
if fext  != '.dxf':
    print(argv[argl-1]+' is not a dxf file')
    exit()
    
lo = pya.LoadLayoutOptions()
layout = pya.Layout()
layout.read(fname+".dxf")

import FreeCAD
import importDXF, Draft

doc = FreeCAD.newDocument()
doc.saveAs(fname+'.FCStd')

layer_names=[]
for li in layout.layer_indexes():
    linfo=layout.get_info(li)
    if linfo.layer!=0 and linfo.name!="_0"  and linfo.name!="L0D0_0":
       layer_names.append(linfo.name)

for name in layer_names:
  body=doc.addObject("PartDesign::Body", name)
  layout.clear()
  layout.read(fname+".dxf")
  for c in layout.each_cell():
      if c.name=="TOP":
          c.flatten(-1,True)
  for c in layout.each_cell():
      if c.name!="TOP":
          layout.delete_cell(c.cell_index)
  for li in layout.layer_indexes():
     linfo=layout.get_info(li)
     if linfo.name!=name and linfo.layer!=0:
        layout.delete_layer(li)
  fname2=fname+"_"+name
  layout.write(fname2+".dxf")
  existings = doc.Objects
  importDXF.insert(fname2+".dxf",doc.Name)
  newObjs = [o for o in doc.Objects if o not in existings]
  for obj in newObjs:
      sketch = Draft.make_sketch(obj, autoconstraints=True)
      doc.removeObject(obj.Name)
      body.Group += [sketch]

doc.save()
FreeCAD.closeDocument(doc.Name)

```