realthunder/FreeCAD #423: Freecad crashes When using the fill command.

hamza00789, 2022-05-21:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US) 

As can be seen in the video free cad crashes when I use the fill command. 

https://user-images.githubusercontent.com/92737149/169657169-c0358838-a4ea-42e3-91c5-cfd964a86d50.mp4  

I know there is some distance between the wires. But freecad should not crash because of this.      

[image]         

[image]  

[image]  

File   

[1.zip](https://github.com/realthunder/FreeCAD/files/8747826/1.zip)

closed 2022-05-24 by realthunder

realthunder, 2022-05-24:
The crash is due to an OCC bug as mentioned in code comment [here](https://github.com/realthunder/FreeCAD/blob/3f41f321039021fd628765164f00ff8ec4f28f2b/src/Mod/Part/App/TopoShapeEx.cpp#L4171).

BTW, with commit 8916f67e4cb20557dfa7e12f551fe2ecec44c7bc, it is now possible to create this kind of filled face with the green `SubShapeBinder`, if you don't need any extra face or edge constrain.

https://user-images.githubusercontent.com/1207888/169925049-2258c4ec-206e-4181-ad11-e4f7851e3461.mp4

realthunder, 2022-05-25:
Just for reference. There will be another crash if creating a (non-filled) surface using the edges of the existing face in the test file.

[image]

hamza00789, 2022-05-26:
Coming this distance is also a problem. 
The distance should not have come to this place. 
 
[image]
 
[distance 1.zip](https://github.com/realthunder/FreeCAD/files/8775523/distance.1.zip)

commit 3f41f321 (2022-05-24): Surface: refactor Filling

Refactor using Part::TopoShape::makEFilledFace().
Fix crash due to OCC bug.

Fixes #423
@hamza00789