realthunder/FreeCAD #424: Open FreeCAD file using "open with" option - doesn't work

bambukouk, 2022-05-24:
Snap version, as per forum post:
https://forum.freecadweb.org/viewtopic.php?f=42&t=65662&p=597126#p597098
[image]
[image]

Problem is caused by `%F` missing from .desktop file
adding it solves the problem:
[image]
```
OS: Ubuntu Core 20 (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.520.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.05.20-edge)
Hash: d47fafeec334a1069256639f391f17368276068c
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United Kingdom (en_GB)
```

luzpaz, 2022-05-24:
Want to make a PR ? :smirk:

bambukouk, 2022-05-24:
> Want to make a PR ? smirk

It works fine in master branch... 
I am suggesting it needs to be corrected in Link branch so it also works OK out of the box.
One wouldn't expect to have to mess with innards of snap?
It only works for me because I use copied .desktop (outside of snap) and modify it myself to customise integration in OS.
as for PR... I don't even know what it is :+1:

ppd, 2022-05-24:
The desktop file for the realthunder snap should be this one: https://github.com/realthunder/FreeCADMakeImage/blob/master/conda/branding/asm3/AppDir/freecad_link.desktop

bambukouk, 2022-05-26:
thank you @ppd 👍

closed 2022-05-26 by realthunder