realthunder/FreeCAD #426: Refine is not working.

hamza00789, 2022-05-24:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)

The angle of extrude and cone has become one, the refine should work. As can be seen in the file refine did not work.  

[File.zip](https://github.com/realthunder/FreeCAD/files/8765978/File.zip)

[image]

hamza00789, 2022-05-25:
related issue  #336

hamza00789, 2022-05-26:
> cone angle is 11.3099deg extrude angle is 11.31deg refine will never work with these numbers...     

How did you find the angle value of the cone? 11.3099deg     
I think the Cone angle is 11.31deg.   
     
[image]    

Also, it is impossible to give such an angle to the extrude.11.3099deg 

[image]

bambukouk, 2022-05-26:
> How did you find the angle value of the cone? 11.3099deg I think the Cone angle is 11.31deg.

simple trigonometry...
tan-1 (2/10)=11.30993247 degrees
not surprisingly FreeCAD agrees 😄
sketch simple triangle and find derived angular dimension