realthunder/FreeCAD #428: single point from sketch remains invisible outside of sketch.

jx2014, 2022-05-25:
[invisible_single_points.zip](https://github.com/realthunder/FreeCAD/files/8773056/invisible_single_points.zip)

Steps to create this bug:
1. create a sketch
2. create some points
3. make those points normal (by default they are 'construction points')
4. go out of sketch
5. those points remain invisible in Linkstage 2022.111.28429.
6. In Official FreeCad release rsion: 0.19.24267 +148 (Git), those points are visible outside of the sektch

Original bug description in Freecad community forum: [https://forum.freecadweb.org/viewtopic.php?f=3&t=68926&p=597389#p597389](url)

[image]

[image]

> OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

However, in freecad, the single point is visible outside the sketch

[image]

> OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

bambukouk, 2022-05-26:
Your link to forum post doesn't work,
it's better this way:
https://forum.freecadweb.org/viewtopic.php?f=3&t=68926&p=597389#p597389

Shkolik, 2022-07-09:
Still reproducible in:

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)