realthunder/FreeCAD #429: External geometry Issue

hamza00789, 2022-05-26:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US) 

External geometry problem. As can be seen in the video, external geometry does not come into the right place.  

https://user-images.githubusercontent.com/92737149/170491007-00bef621-cacf-4d4f-b2aa-ea74c4129be5.mp4

[File.zip](https://github.com/realthunder/FreeCAD/files/8779061/File.zip)

bambukouk, 2022-05-27:
It's the `Group` that seems to be confusing FreeCAD, remove it and all is fine:
[image]

realthunder, 2022-05-28:
Duplicate to #391, and has been fixed.

closed 2022-05-28 by realthunder

bambukouk, 2022-05-28:
so... I guess I didn't have problem because I was using latest snap?
Should have given FreeCAD version details in my reply - sorry.