realthunder/FreeCAD #430: Additive Pipe Multisection GUI problem in Part Design WB

projectcad, 2022-05-26:
Multisection option of Additive Pipe GUI doesnt allow to add an additional section, by the way it can be added via properties menu..

One more thing, please let me know which source file /files for setting font/font size of main window of freecad

realthunder, 2022-05-27:
I can't seem to reproduce the problem. Can you please record a screencast showing the problem.

https://user-images.githubusercontent.com/1207888/170629792-3a191911-ba33-42e2-98fa-60258b30f444.mp4

The source code for font setting is [here](https://github.com/realthunder/FreeCAD/blob/add38c8be4a7b75f1f290817f93bf72b6466148b/src/Gui/ViewParams.cpp#L6221). It is a callback in response to change of a configuration parameter `BaseApp/Preferences/View/DefaultFontSize`

projectcad, 2022-05-28:
I just become aware of that difference of selection between "sweep path" and "Multisection option" while adding section missleaded me.

due to,

-at sweep path 

selection can be done after pressing "object" or "add edge"

by the way,

-at Multisection option

it becomes ready to select after changing "constant" to "Multisection"

Thanks for your interest...

projectcad, 2022-05-29:
Maybe selection status should be equalized

closed 2022-05-29 by projectcad