realthunder/FreeCAD #433: Quit menu item always disabled

horfee, 2022-05-30:


luzpaz, 2022-05-30:
Can you recreate on LinkDaily

horfee, 2022-05-30:
Reproduced on this release (the latest available) : 

OS: macOS 12.3
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: C/Default (C)

bambukouk, 2022-05-30:
Is this the same as `Exit (Alt+F4)` ??
I guess it must be mac specific?

my version:
```
OS: Ubuntu Core 20 (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 2022.527.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.05.27-edge)
Hash: 39c9a013e168a7ebb2f8cb2241de144bb0b20b5d
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United Kingdom (en_GB)

```

horfee, 2022-05-30:
Command+Q is the same as Alt+F4 on windows you are right.
This is the standard shortcut for terminating any app on a mac.

notthewave, 2022-11-01:
works fine for me

luzpaz, 2022-11-09:
Closing. Please re-open if persists

closed 2022-11-09 by luzpaz