realthunder/FreeCAD #443: [feature request] Icons in tooltip?

MisterMakerNL, 2022-06-08:
Was working on the icon theme's and had some issues into making the icons readable and understandable.
Then I realized that there is a tooltip and we can just add a big icon too the tooltip, and you can just hoover over it and get a good look at it. 
Like so:
[image]
Also looked around and found out it's possible with some simple code:
https://stackoverflow.com/questions/34296100/use-a-picture-or-image-in-a-qtooltip
Although I wonder if for this every tooltip needs to be manually edited with an image or that the tooltip can pull the icon from the feature.
The stack overflow talks about "pulling it from memory" but I have no idea what they means, maybe you can have a look at it!?
Forum topic:
https://forum.freecadweb.org/viewtopic.php?f=34&t=69264

closed 2022-06-11 by realthunder

realthunder, 2022-06-11:
It is implemented with support for both static icon and GIF animation. See the commit comment above for details.

https://user-images.githubusercontent.com/1207888/173191028-629f0273-cac6-4e53-9c84-c2cc325da5c2.mp4

MisterMakerNL, 2022-06-11:
Love it!! Thank you!

commit 32522009 (2022-06-11): Gui: support icon in tooltip of tool bar button and menu item

Add option Preference -> General -> ToolTipIconSize to control static
icon size in tooltip. Set to zero to disable icon in tooltip.

If enabled, it also support GIF animation if there exists a GIF file at
the same location as the icon file of the tool buton or menu. For
example, for a tool button icon file of resource file address

    iconset:MyIcons/PartDesign_Body.svg

It tries to look for a GIF file with path

    iconset:MyIcons/PartDesign_Body.svg.gif

If found, the GIF image or animation will be displayed in tool tip in
full [... cut]