realthunder/FreeCAD #444: FreeCAD gives illegal storage access when you check the Navigation Cube on.

YasserRoshdi, 2022-06-08:
Make New file then Edit>Reference>Display>Navigation>check Navigation cube on
as soon as the mouse enter graphic working area it gives the message : FreeCAD main executable has stopped working
then if i restart freecad and checked it off (keeping mouse away from the working area till i make the change) everything goes fine.
OS: Windows 7 64 bit service pack 1.
FreeCAD version : 2202.430
Revison No. 28641 (git)
[image]

realthunder, 2022-06-09:
When the illegal storage dialog happen, can you please go to your user directory (`C:\Users\<username>\AppData\Roaming\FreeCAD`) and find the file `crash.dmp`. Zip it and post it here?

YasserRoshdi, 2022-06-09:
Here are both crash.dmp and crash.log
[Crash.zip](https://github.com/realthunder/FreeCAD/files/8867959/Crash.zip)

YasserRoshdi, 2022-06-19:
Sorry for being late, i just saw it now
the same issue exist
here is all freecad folder created by your release 11/5
[FreeCAD.zip](https://github.com/realthunder/FreeCAD/files/8934564/FreeCAD.zip)

realthunder, 2022-06-19:
Hi, looks like this is related to the fact that Win7 is end of support. Please check out [this](https://forum.freecadweb.org/viewtopic.php?style=5&f=22&t=65037#p565117)post for possible fix suggested by others.

YasserRoshdi, 2022-06-19:
Well, it should be a message telling users that FreeCAD is not supported any more on Window 7.
and for the error, i can guess that it is something related to the opengl version.
anyhow thank you for your concerning.

MisterMakerNL, 2022-07-10:
I don't think there is any latest release of any CAD program that still support Windows 7. Freecad runs great on Linux sooooo ;)

discip, 2022-10-05:
@YasserRoshdi
Please consider closing this issue, since there will not be any official solution in the foreseeable future. 😊