realthunder/FreeCAD #445: Assert fail const SbVec3f& SoCoordinateElement::get3(int) index == 1 and this->numCoords == 1

jharvell, 2022-06-09:
I have a simple file with only a slab and a few walls created using BIM Workbench.  Will add link to file if that will help.

When I load the file, I see the following output on the terminal:
<TopoShape> ArchSite.py(1139)|TopoShape.cpp(1110): failed to get bounding box: Bnd_Box is void
<TopoShape> ArchSite.py(1139)|TopoShape.cpp(1110): failed to get bounding box: Bnd_Box is void
Coin warning in [indexedlineset]::GLRender(): Erroneous coordinate index: 1 (Should be within [0, 0]). Aborting rendering. This message will be shown once, but there might be more errors

Then when I set the view to Top and move the cursor to the bottom left corner of one of the walls, FreeCAD asserts as follows:
FreeCAD: /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/elements/SoCoordinateElement.cpp:179: const SbVec3f& SoCoordinateElement::get3(int) const: Assertion `index >= 0 && index < this->numCoords' failed.
Aborted (core dumped)

GDB shows the following:
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007f97d8be55cf in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007f97d8b998a2 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007f97d8b84469 in __GI_abort () at abort.c:79
#4  0x00007f97d8b84395 in __assert_fail_base (fmt=<optimized out>, assertion=<optimized out>, file=<optimized out>, line=<optimized out>, function=<optimized out>)
    at assert.c:92
#5  0x00007f97d8b92b12 in __GI___assert_fail
    (assertion=0x7f97dbd707e8 "index >= 0 && index < this->numCoords", file=0x7f97dbd70698 "/var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/elements/SoCoordinateElement.cpp", line=179, function=0x7f97dbd707b0 "const SbVec3f& SoCoordinateElement::get3(int) const") at assert.c:101
#6  0x00007f97db916a53 in SoCoordinateElement::get3(int) const (this=0x55b38a037ca0, index=1)
    at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/elements/SoCoordinateElement.cpp:179
#7  0x00007f97dbb87320 in SoIndexedLineSet::generatePrimitives(SoAction*) (this=0x55b389765160, action=0x55b389194310)
    at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/shapenodes/SoIndexedLineSet.cpp:726
#8  0x00007f97dbbbaa40 in SoShape::rayPick(SoRayPickAction*) (this=0x55b389765160, action=0x55b389194310)
    at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/shapenodes/SoShape.cpp:459
#9  0x00007f9754037225 in PartGui::SoBrepEdgeSet::rayPick(SoRayPickAction*) (this=0x55b389765160, action=0x55b389194310)
    at /mnt/etretat/home/jharvell/git/FreeCAD/src/Mod/Part/Gui/SoBrepEdgeSet.cpp:653
#10 0x00007f97df305d9f in Gui::SoFCRayPickAction::doPick(SoNode*) (this=0x55b389194310, node=0x55b389765160)
    at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/SoFCSelectionAction.cpp:1643
#11 0x00007f97df3055c4 in operator()(SoAction*, SoNode*) const (__closure=0x0, action=0x55b389194310, node=0x55b389765160)
    at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/SoFCSelectionAction.cpp:1559
#12 0x00007f97df305682 in _FUN(SoAction*, SoNode*) () at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/SoFCSelectionAction.cpp:1564
#13 0x00007f97db7ed9a5 in SoAction::traverse(SoNode*) (this=0x55b389194310, node=0x55b389765160)
    at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/actions/SoAction.cpp:950
#14 0x00007f97dba175f5 in SoChildList::traverse(SoAction*, int, int) (this=0x55b38975bf70, action=0x55b389194310, first=0, last=7)
    at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/misc/SoChildList.cpp:357
#15 0x00007f97dba1781f in SoChildList::traverse(SoAction*) (this=0x55b38975bf70, action=0x55b389194310)
    at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/misc/SoChildList.cpp:427
#16 0x00007f97dbb250e6 in SoGroup::doAction(SoAction*) (this=0x55b3897698c0, action=0x55b389194310)
    at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/nodes/SoGroup.cpp:548
[...]
#99 0x00007f97d5e9006f in g_main_context_iteration (context=0x7f97cc005010, may_block=1) at ../glib-2.70.0/glib/gmain.c:4240
#100 0x00007f97da2130f8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=0x55b386c6e180, flags=...) at /var/tmp/portage/dev-qt/qtcore-5.15.2-r10/work/qtbase-a4f9e56975fa6ab4a1f63a9b34a4d77b1cfe4acd/src/corelib/kernel/qeventdispatcher_glib.cpp:423
#101 0x00007f97da1c1093 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (this=0x7fff0ae87900, flags=...) at ../../include/QtCore/../../../qtbase-a4f9e56975fa6ab4a1f63a9b34a4d77b1cfe4acd/src/corelib/global/qflags.h:69
#102 0x00007f97da1cc490 in QCoreApplication::exec() () at ../../include/QtCore/../../../qtbase-a4f9e56975fa6ab4a1f63a9b34a4d77b1cfe4acd/src/corelib/global/qflags.h:121
#103 0x00007f97dec17500 in Gui::Application::runApplication() () at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/Application.cpp:2564
#104 0x000055b385c2fa64 in main(int, char**) (argc=1, argv=0x7fff0ae884c8) at /mnt/etretat/home/jharvell/git/FreeCAD/src/Main/MainGui.cpp:293
(gdb) f 6
#6  0x00007f97db916a53 in SoCoordinateElement::get3 (this=0x55b38a037ca0, index=1)
    at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/elements/SoCoordinateElement.cpp:179
179       assert(index >= 0 && index < this->numCoords);
(gdb) p *this
$1 = {<SoReplacedElement> = {<SoElement> = {_vptr.SoElement = 0x7f97dbf7f9a8 <vtable for SoCoordinateElement+16>, static classStackIndex = -1, typeId = {
        index = 158}, stackIndex = 34, static stackToType = 0x55b387261310, depth = 6, static classTypeId = {index = 112}, nextup = 0x55b38d423b40, 
      nextdown = 0x55b38d693290}, static classStackIndex = -1, static classTypeId = {index = 157}, nodeId = 81666}, static classStackIndex = 34, 
  static classTypeId = {index = 158}, numCoords = 1, coords3D = 0x55b38976a0b0, coords4D = 0x0, areCoords3D = 1, dummy3D = {vec = {0, 1.821688e-44, 0}}, dummy4D = {
    vec = {0, 0, 0, 0}}, static initialdefaultcoords = 0x55b387284410}

FreeCAD version:
jharvell@wolfhound FreeCAD$ git --no-pager log -n1
commit 8cbb87298dc6ddeb32791d847317adbf3b6e800e (HEAD -> LinkDaily, tag: 2022.06.07-edge, link/LinkDaily)
Author: Zheng, Lei <realthunder.dev@gmail.com>
Date:   Tue Jun 7 22:35:49 2022 +0800

    PartDesign: fix Loft/Pipe editing task spine selection

realthunder, 2022-06-09:
Please post your file here, either zip it, or change the extension to zip. BTW, are you compiling libCoin3D yourself? If so, which revision are you using?

jharvell, 2022-06-10:
file:///tmp/mink.FCStd.zip

jharvell, 2022-06-10:
[mink.FCStd.zip](https://github.com/realthunder/FreeCAD/files/8874901/mink.FCStd.zip)

File attached. libCoin3D is from Gentoo package media-libs/coin-4.0.0.  I did do a custom build of it through Gentoo's build system to remove all optimization so I could see the values related to the assert that were otherwise optimized out.  The Gentoo package seems to be getting the source from https://github.com/coin3d/coin/releases/download/coin-4.0.0/coin/Coin/coin.4.0.0-src.tar.gz

closed 2022-06-13 by realthunder

jharvell, 2022-06-13:
I get the same crash with your fix.  Also, SoFCRenderCacheManagerP::SoFCRenderCacheManagerP() is not in the stack trace.  Here's the whole stack trace:

(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007fb4086e15cf in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007fb4086958a2 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007fb408680469 in __GI_abort () at abort.c:79
#4  0x00007fb408680395 in __assert_fail_base (fmt=<optimized out>, assertion=<optimized out>, file=<optimized out>, line=<optimized out>, function=<optimized out>) at assert.c:92
#5  0x00007fb40868eb12 in __GI___assert_fail
    (assertion=0x7fb40b86c7e8 "index >= 0 && index < this->numCoords", file=0x7fb40b86c698 "/var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/elements/SoCoordinateElement.cpp", line=179, function=0x7fb40b86c7b0 "const SbVec3f& SoCoordinateElement::get3(int) const") at assert.c:101
#6  0x00007fb40b412a53 in SoCoordinateElement::get3(int) const (this=0x55aab3cff3e0, index=1) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/elements/SoCoordinateElement.cpp:179
#7  0x00007fb40b683320 in SoIndexedLineSet::generatePrimitives(SoAction*) (this=0x55aab1ab2290, action=0x55aab0297750) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/shapenodes/SoIndexedLineSet.cpp:726
#8  0x00007fb40b6b6a40 in SoShape::rayPick(SoRayPickAction*) (this=0x55aab1ab2290, action=0x55aab0297750) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/shapenodes/SoShape.cpp:459
#9  0x00007fb3840c9285 in PartGui::SoBrepEdgeSet::rayPick(SoRayPickAction*) (this=0x55aab1ab2290, action=0x55aab0297750) at /mnt/etretat/home/jharvell/git/FreeCAD/src/Mod/Part/Gui/SoBrepEdgeSet.cpp:653
#10 0x00007fb40ee175af in Gui::SoFCRayPickAction::doPick(SoNode*) (this=0x55aab0297750, node=0x55aab1ab2290) at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/SoFCSelectionAction.cpp:1643
#11 0x00007fb40ee16dd4 in operator()(SoAction*, SoNode*) const (__closure=0x0, action=0x55aab0297750, node=0x55aab1ab2290) at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/SoFCSelectionAction.cpp:1559
#12 0x00007fb40ee16e92 in _FUN(SoAction*, SoNode*) () at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/SoFCSelectionAction.cpp:1564
#13 0x00007fb40b2e99a5 in SoAction::traverse(SoNode*) (this=0x55aab0297750, node=0x55aab1ab2290) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/actions/SoAction.cpp:950
#14 0x00007fb40b5135f5 in SoChildList::traverse(SoAction*, int, int) (this=0x55aab1bbe910, action=0x55aab0297750, first=0, last=7) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/misc/SoChildList.cpp:357
#15 0x00007fb40b51381f in SoChildList::traverse(SoAction*) (this=0x55aab1bbe910, action=0x55aab0297750) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/misc/SoChildList.cpp:427
#16 0x00007fb40b6210e6 in SoGroup::doAction(SoAction*) (this=0x55aab1bc02a0, action=0x55aab0297750) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/nodes/SoGroup.cpp:548
#17 0x00007fb40b644cb9 in SoSeparator::doAction(SoAction*) (this=0x55aab1bc02a0, action=0x55aab0297750) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/nodes/SoSeparator.cpp:487
#18 0x00007fb40b645f3b in SoSeparator::rayPick(SoRayPickAction*) (this=0x55aab1bc02a0, action=0x55aab0297750) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/nodes/SoSeparator.cpp:862
#19 0x00007fb40b62d445 in SoNode::rayPickS(SoAction*, SoNode*) (action=0x55aab0297750, node=0x55aab1bc02a0) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/nodes/SoNode.cpp:1124
#20 0x00007fb40b2e99a5 in SoAction::traverse(SoNode*) (this=0x55aab0297750, node=0x55aab1bc02a0) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/actions/SoAction.cpp:950
#21 0x00007fb40b5135f5 in SoChildList::traverse(SoAction*, int, int) (this=0x55aab1620ea0, action=0x55aab0297750, first=0, last=0) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/misc/SoChildList.cpp:357
#22 0x00007fb40b51388d in SoChildList::traverse(SoAction*, int) (this=0x55aab1620ea0, action=0x55aab0297750, index=0) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/misc/SoChildList.cpp:438
#23 0x00007fb40b64b0c7 in SoSwitch::doAction(SoAction*) (this=0x55aab1620cd0, action=0x55aab0297750) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/nodes/SoSwitch.cpp:315
#24 0x00007fb40edbb082 in SoFCSwitch::doAction(SoAction*) (this=0x55aab1620cd0, action=0x55aab0297750) at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/Inventor/SoFCSwitch.cpp:213
#25 0x00007fb40edbbb14 in SoFCSwitch::pick(SoPickAction*) (this=0x55aab1620cd0, action=0x55aab0297750) at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/Inventor/SoFCSwitch.cpp:419
#26 0x00007fb40b62d345 in SoNode::pickS(SoAction*, SoNode*) (action=0x55aab0297750, node=0x55aab1620cd0) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/nodes/SoNode.cpp:1098
#27 0x00007fb40b2e99a5 in SoAction::traverse(SoNode*) (this=0x55aab0297750, node=0x55aab1620cd0) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/actions/SoAction.cpp:950
#28 0x00007fb40b5135f5 in SoChildList::traverse(SoAction*, int, int) (this=0x55aab16208d0, action=0x55aab0297750, first=0, last=1) at /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/misc/SoChildList.cpp:357
[... cut ...]
#101 0x00007fb409cbd093 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (this=0x7ffc45180ff0, flags=...) at ../../include/QtCore/../../../qtbase-a4f9e56975fa6ab4a1f63a9b34a4d77b1cfe4acd/src/corelib/global/qflags.h:69
#102 0x00007fb409cc8490 in QCoreApplication::exec() () at ../../include/QtCore/../../../qtbase-a4f9e56975fa6ab4a1f63a9b34a4d77b1cfe4acd/src/corelib/global/qflags.h:121
#103 0x00007fb40e72555e in Gui::Application::runApplication() () at /mnt/etretat/home/jharvell/git/FreeCAD/src/Gui/Application.cpp:2572
#104 0x000055aaad077a64 in main(int, char**) (argc=1, argv=0x7ffc45181bb8) at /mnt/etretat/home/jharvell/git/FreeCAD/src/Main/MainGui.cpp:293
(gdb)

realthunder, 2022-06-14:
Right. The referenced commit does not really address the same problem as the one you reported here, but it is the same root cause, which is that Arch::BuildingPart set some out of bound index in some libCoin3D node. The debug build of libCoin3D will check for that in the form of an assertion, but the release build works just fine, indicating that there are other safety check in libCoin3D to deal with this situation. So I think it's good enough, although it would be better to make it a soft warning even in debug build rather than a hard assertion failure. But then again, libCoin3D is an external library. With FreeCAD being an open system, any external workbench is free to add its customized libCoin3D nodes, there is no way to prevent this kind of mistake in node setup.

On the other hand, the reason I add the sanity check in the referenced commit is because the partial geometry information obtained from this kind of faulty node produces visual defects in my experimental render caching.

commit a541a498 (2022-06-13): Renderer: add sanity check for SoIndexedShape

Fixes #445
@jharvell