realthunder/FreeCAD #447: FreeCAD LinkStage AppImage crashes on startup

VorpalBlade, 2022-06-13:
I haven't started FreeCAD LinkStage for a while on this computer but now it crashes on startup after the loading screen is completed. 0.19 starts fine. My first step was to upgrade to the last nightly version but the issue kept happening. 

* OS is Arch Linux.
* Graphics is Intel with Mesa drivers.
* I'm using the AppImage of LinkStage. I'm using the distro package for 0.19.
* Version is FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220430-glibc2.12-x86_64.AppImage (but it also happened before I updated, unfortunately I did not make a note of what version I had then). I also downloaded the stable version. It crashed in the same way.
* On my desktop with nvidia graphics (also running Arch Linux), there is no crash.

Console log:
```
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 475, resource id: 14758470, major code: 40 (TranslateCoords), minor code: 0
/tmp/.mount_FreeCAptUUQ9
FreeCAD 0.20, Libs: 2022.430R26244 +4758 (Git)
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAptUUQ9/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAptUUQ9/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAptUUQ9/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
1.317154 <asm3.main> init_gui.py(14): no solver backend found
/tmp/.mount_FreeCAptUUQ9/AppRun: line 46: 25916 Aborted                 (core dumped) "$PREFIX/bin/$name" "$@"
```

It seems plausible that the mesa warnings are relevant. If so it seems that the appimage contains a too old version of libstdc++, and it would be required that the system version of libstdc++ is used (or the appimage bundles copies of graphics drivers).

To get a backtrace from the core dump I mounted the AppImage (--appimage-mount) and ran with gdb and got the following backtrace:

```
Thread 1 "FreeCADLink" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
Downloading 0.00 MB source file /usr/src/debug/glibc/nptl/pthread_kill.c
44	     return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;                                                                        
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff46823d3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff4632838 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff461c535 in __GI_abort () at abort.c:79
#4  0x00007ffff489035b in QMessageLogger::fatal(char const*, ...) const () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Core.so.5
#5  0x00007fffea772ddf in ?? () from /tmp/.mount_FreeCAJtHBba/usr/plugins/designer/../../lib/libQt5QuickWidgets.so.5
#6  0x00007fffea772f53 in ?? () from /tmp/.mount_FreeCAJtHBba/usr/plugins/designer/../../lib/libQt5QuickWidgets.so.5
#7  0x00007fffea774d57 in QQuickWidget::resizeEvent(QResizeEvent*) () from /tmp/.mount_FreeCAJtHBba/usr/plugins/designer/../../lib/libQt5QuickWidgets.so.5
#8  0x00007fffea74dcb9 in ?? () from /tmp/.mount_FreeCAJtHBba/usr/plugins/designer/../../lib/libQt5WebEngineWidgets.so.5
#9  0x00007ffff599f982 in QWidget::event(QEvent*) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#10 0x00007fffea77589b in QQuickWidget::event(QEvent*) () from /tmp/.mount_FreeCAJtHBba/usr/plugins/designer/../../lib/libQt5QuickWidgets.so.5
#11 0x00007fffea74e02e in ?? () from /tmp/.mount_FreeCAJtHBba/usr/plugins/designer/../../lib/libQt5WebEngineWidgets.so.5
#12 0x00007ffff59747f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#13 0x00007ffff597a31d in QApplication::notify(QObject*, QEvent*) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#14 0x00007ffff6f6b5f9 in Gui::GUIApplication::notify(QObject*, QEvent*) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libFreeCADGui.so
#15 0x00007ffff49ad606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Core.so.5
#16 0x00007ffff599a56f in QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#17 0x00007ffff599d121 in QWidgetPrivate::show_helper() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#18 0x00007ffff599d0ee in QWidgetPrivate::showChildren(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#19 0x00007ffff599d13d in QWidgetPrivate::show_helper() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#20 0x00007ffff599f23b in QWidgetPrivate::setVisible(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#21 0x00007ffff599d0f8 in QWidgetPrivate::showChildren(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#22 0x00007ffff599d13d in QWidgetPrivate::show_helper() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#23 0x00007ffff599d0ee in QWidgetPrivate::showChildren(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#24 0x00007ffff599d13d in QWidgetPrivate::show_helper() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#25 0x00007ffff599d0ee in QWidgetPrivate::showChildren(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#26 0x00007ffff599d13d in QWidgetPrivate::show_helper() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#27 0x00007ffff599f23b in QWidgetPrivate::setVisible(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#28 0x00007ffff599d0f8 in QWidgetPrivate::showChildren(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#29 0x00007ffff599d13d in QWidgetPrivate::show_helper() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#30 0x00007ffff599f23b in QWidgetPrivate::setVisible(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#31 0x00007ffff599d0f8 in QWidgetPrivate::showChildren(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#32 0x00007ffff599d13d in QWidgetPrivate::show_helper() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#33 0x00007ffff599f23b in QWidgetPrivate::setVisible(bool) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libQt5Widgets.so.5
#34 0x00007ffff7503c19 in Gui::MainWindow::loadWindowSettings() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libFreeCADGui.so
#35 0x00007ffff6ed96db in Gui::postMainWindowSetup(Gui::MainWindow&) () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libFreeCADGui.so
#36 0x00007ffff6eda279 in Gui::Application::runApplication() () from /tmp/.mount_FreeCAJtHBba/usr/bin/../lib/libFreeCADGui.so
#37 0x000055555555870d in main ()
```

ngocminhdao88, 2022-06-14:
I got the same problem with the AppImage crashing, because of Mesa graphic driver and libstdc++.

The libstdc++ packaged in AppImage ver 4.30 doesn't support GLIBCXX_3.4.30. Mesa driver loader requires it.

You can try this (please fix the path and your appimage's name)
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./FreeCAD_weekly-builds-27936-Linux-Conda_glibc2.12-x86_64.AppImage`

Another solution is extract the appimage and copy the system's libstdc++ over. Symbolic link it to libstdc++.so.6 and libstdc++.so.29. This is what i did and it works.

discip, 2022-10-05:
@VorpalBlade
If this is fixed, than please close this issue. 😊

VorpalBlade, 2022-10-06:
Yes, this appears to be fixed.