realthunder/FreeCAD #449: SubShapeBinder in wrong position if parent assembly is linked and moved somwhere

NomAnor, 2022-06-16:
### Bug Description
I have an assembly (Assembly 0) that has a body (In-Context) with a SubShapeBinder referencing a face of another part. This works so far.

If I create a Link of the assembly and move that link to another position, the position of the SubShapeBinder is updated according to the position of the link and not the original assembly.

[image]

The yellow SubShapeBinder is the one in the original assembly. The selected green one is in the link. The linked shape is in the lower right corner, the original one ist above it.

This will obviously break any feature referencing the binder shape (the yellow original one). E.g. I want the pad to extrude to the face of the binder which will not work because it is now in the wrong position.

### Full version info
```
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.31002 (Git)
Build type: Release
Branch: makepkg
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.10.5
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)
```

Build from commit 68f794119867077c3cf7b2e2187b81c040f04053

### Test file
[freecad_binder.zip](https://github.com/realthunder/FreeCAD/files/8918749/freecad_binder.zip)

NomAnor, 2022-06-16:
The direct cause seems to be that the `Context` property is wrongly set to

```
>>> obj.Context
(<App::Link object>, 'Assembly.Parts.Body002.Binder.')
```
making the link coordinate system the base for the position calculation.

I think the property is set based on the `Parents` list that is based on the `InList`.

```
>>> obj.Parents
[(<App::Link object>, 'Parts.Body002.Binder.'), (<App::Link object>, 'Assembly.Parts.Body002.Binder.')]
```
This should show all possible parents.

The first one seems correct in a way (This should not be the first entry, that should be the "real" Parent). Looking at the tree view, the link is a parent of the Binder (The copy of it).

But the second one is wrong. It looks like the convention is that an empty parent list means the document is the parent. But in this case the assembly is both the child of the document and the child of the link, but the document is not included (can't be) in the parent list:

```
>>> assembly.Parents
[(<App::Link object>, 'Assembly.')]
```

I'm not sure if this is correct but maybe it helps.

NomAnor, 2022-06-17:
Putting the Assembly 0 inside a Part oder another Assembly and recomputing the binder solves the issue. Putting it inside a Group does not.

I encountered this when I created a couple small assemblies in the document root and then used them as links inside another root assembly in the same file.

NomAnor, 2022-06-17:
Playing around a bit more, putting Assembly 0 inside a Part only works if it sits at the origin of the part. If I change the placement of the assembly the binder has the wrong position again.

Reverting commit 688368aac seems to solve it even when I vary the placement of the assembly inside the part. So the Context is not the problem but the `resolveRelativeLink()` call inside the matrix loop.

realthunder, 2022-06-18:
The problem should be fixed in the above commit. Are you building FreeCAD yourself? If so, maybe you can test it more. Thanks!

closed 2022-06-18 by realthunder

NomAnor, 2022-06-18:
I compiled the new code and so far it works

commit bd42d5ca (2022-06-18): App: fix DocumentObject::resolveRelativeLink() with option TopParent

Fixes #449
@NomAnor