realthunder/FreeCAD #451: Revolving sketch around one of its edges fails

antonok-edm, 2022-06-17:
## Steps to reproduce

1. Create a new part in the Part Design workbench
2. Create a new sketch on the XY axis
3. Use the rectangle tool to draw an arbitrary rectangle with one corner at the origin of the sketch
4. Use the `Revolution` tool to revolve the sketch around one of the axes - `Vertical sketch axis`, `Horizontal sketch axis`, `Base Y axis`, or `Base X axis`. Default settings otherwise.
5. The operation will fail with the error `gp_Dir::Crossed() - result vector has zero norm`

[image]

This error seems to occur in any orientation as long as the axis of rotation coincides with an edge of the sketch.

### Version info

I built this FreeCAD version using the `freecad-linkstage3-git` Arch Linux AUR package, from git commit 22643458936e013c38a8d7069aede81b5bbe42fa.

realthunder, 2022-06-19:
I can't reproduce the problem. Can you please post your file here? Also, please post your complete FreeCAD About information here? Maybe you are using some outdated OCC version.

Bonnee, 2022-06-25:
I'm having the same problem when revolving a sketch on any of the body axis

antonok-edm, 2022-06-27:
@realthunder sorry for the delay :sweat_smile:

I actually discovered this when trying to modify parameters of the default endmill tool from the Path workbench, so that model should reproduce the issue (i.e. `/usr/lib/freecad/Mod/Path/Tools/Shape/endmill.fcstd`)

Here's my FreeCAD About information:

```
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.20.31135 (Git)
Build type: Release
Branch: makepkg
Hash: 22643458936e013c38a8d7069aede81b5bbe42fa
Python version: 3.10.5
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

OCC in particular is the default `community/opencascade` package from the Arch Linux repos.

antonok-edm, 2022-07-06:
@realthunder Not 100% sure what caused it, but between the most recent changes on the branch and [updates to the AUR package](https://aur.archlinux.org/cgit/aur.git/log/?h=freecad-linkstage3-git), I'm no longer able to reproduce the issue. Thanks for looking and sorry about the noise!

@Bonnee updating your `freecad-linkstage3-git` package should fix it!

closed 2022-07-06 by antonok-edm

Bonnee, 2022-07-06:
I'm still getting the same issue on the latest `freecad-linkstage3-bin`. I'll try bumping the PKGBUILD to the latest `freecad-linkstage3-git` and see if it makes any difference

Bonnee, 2022-07-06:
I can confirm that the issue disappeared with the latest `freecad-linkstage3-git` PKGBUILD.