realthunder/FreeCAD #452: [Feature Request] Sketcher copy/paste (via xml) included in Link branch (see upstream PR)

derMart, 2022-06-18:
Hey, it seems the very useful feature of generic copy paste sketcher geometries in between different sketches, even in between different instances of freecad (see here: https://www.youtube.com/watch?v=ZH432giUasM) is not working in this fork.
If I select a portion of a sketch inside sketcher, the Edit -> Copy menu is greyed out.

I am using FreeCad Link Branch daily snap package (Version 2022.617)

derMart, 2022-06-18:
Yea, sorry, I just found out that pull request. Probably better to merge it upstream.

luzpaz, 2022-06-18:
@derMart feel free to participate in the discussion thread associated with that issue. I think for now, we should close this ticket until upstream adds it? Don't you think? Since Realthunder has is dealing with toponaming and other issues right now.

derMart, 2022-06-18:
sure, you are right

closed 2022-06-18 by derMart