realthunder/FreeCAD #454: Appimage not rebuilt since May 22nd

eMPee584, 2022-06-20:
Hi & thanks for your work.. we are in the process of laying finishing touches on our https://github.com/fsfw-dresden/usb-live-linux and include the latest of your appimage builds, but there hasn't been a new one since May although snaps have been updated and tags have been tagged.. Would be amazing to be able to include a post-0.20-release as appimage (which is slightly more light-weight than snaps)..

realthunder, 2022-06-21:
I plan to make a new release this week. But, the release will not be synced with the latest 0.20 upstream (last sync around Nov last year). I'll do a merge with the upstream after the release and then make a new one, which will take about two weeks if things goes well. But then again, due to the significant changes in upstream in recent months, the _after merge_ release is expected to have many problems and will only get stable after a few iterations.

eMPee584, 2022-06-21:
Ah, thanks a lot for explaining the situation!
Leaves me a bit stretching my head whether I might ship official 0.2o or link branch 2022.04.30.. 🤔

discip, 2022-10-05:
@eMPee584 
Is this still relevant?
If not, please do not forget to close this issue. 😊

closed 2022-10-07 by luzpaz