realthunder/FreeCAD #462: Blank Start Page

QuakeString, 2022-06-30:
There is nothing in start page

OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
[image]

andyylin, 2022-08-08:
> It may be related to [this](https://bugs.launchpad.net/ubuntu/+source/glibc/+bug/1939993) issue. If so, we may need to add a patch to conda qt recipe. @looooo can you please check this out? I see upstream also [reporting](https://forum.freecadweb.org/viewtopic.php?f=4&t=66363&start=60#p579284) similar problem.

I believe so, changing to software OpenGL under Preferences > Display allows the Start page to show.

carsten-br, 2022-08-24:
> I believe so, changing to software OpenGL under Preferences > Display allows the Start page to show.

This apparently works in MacOS, but not with Manjaro KDE. Same issue was reported upstream for MacOS: #6410

looooo, 2022-08-24:
I guess this issue is fixed with qt5.15. I updated the weekly builds and 20.1 (std freecad) to qt 5.15.