realthunder/FreeCAD #468: BIM Door creation issue

QuakeString, 2022-07-01:
Unable to subtract door from a wall
[https://youtu.be/zF187cZFP7E](https://youtu.be/zF187cZFP7E)
above link to demonstrate the issue.
AppImage version 2021.311

The same process I try with FreeCAD 0.19, and there it works as it should.

my system Manjaro (21.0) KDE Plasma 5.21.3

luzpaz, 2022-07-01:
@QuakeString can you identify an open upstream Arch ticket that reflects this ?

yorikvanhavre, 2022-07-04:
Just to clarify, in Arch, windows depend on their "host" wall, not the contrary (Windows have a "Hosts" property that points to their hosts objects. This was made because almost everyone wanted windows position to be dependent on the underlying wall face. A wall dependency on a window should in principle not exist (the wall "soft-scans" for windows attached to it, without creating an explicit dependency).

One should test if the same problem occurs on vanilla FreeCAD...