realthunder/FreeCAD #473: Copy Paste Body into part and Reordering it

wsteffe, 2022-07-04:
Hy RT,
 Sometime I need to  copy a Body placed in a Part into a new Body that I would like to stay in the same part.
But I have observed that the paste always put the body in the document root  outside of the Part even if the Part was selected before. It is possible to drag the copied body into the part but this operation may be tricky if the structure tree is so long that origin and destination of drag are not visible at the same time. It would be more natural if  the pasted object were automatically placed in the pre-selected container.

Another problem I have is related with the position of the pasted Body inside of the containing Part.
After dragging a body into a Part it is always placed at the and of the existing bodies.
I have tried to drag it in a different position but I were never successful with this kind of operation. Most of the time it messed the data structure with the cancellation of an existing body placed near the end point of drag.

wsteffe, 2022-07-04:
Regarding the reordering of Bodies in a Part: would it possible to extend the context command "Move Object After other Object" which works with features in a Body to works also with Bodies in a Part ?

realthunder, 2022-07-04:
Could you please provide a sample project showing the problem you have when reordering a body in a Part with several bodies?

wsteffe, 2022-07-04:
Here it is: [RX_HelicFilter.FCStd.zip](https://github.com/realthunder/FreeCAD/files/9037438/RX_HelicFilter.FCStd.zip)
I have just created with copy/paste the Body named helic7 .
It is not very important but I tried (unsuccesfully) to move it upward in the list of bodies  associated with Part named CMP_HelicFilter.

realthunder, 2022-07-04:
Drag and drop provides more context information than `Move object after other`. To be more precise, the command should be `Move object after another in a given parent`.

https://user-images.githubusercontent.com/1207888/177136115-a2c79633-e516-4d49-a7b0-8a96ff2099e6.mp4

commit 6ce03688 (2022-07-09): Gui: support paste into selected geo feature group

Closes #473
@wsteffe
commit a36d952f (2022-07-09): Gui: support paste into selected geo feature group

Closes #473
@wsteffe
closed 2022-07-09 by realthunder