realthunder/FreeCAD #474: Unexpected Crash

wsteffe, 2022-07-04:
[RX_HelicFilter_crashing.FCStd.zip](https://github.com/realthunder/FreeCAD/files/9040045/RX_HelicFilter_crashing.FCStd.zip). FC crash on this file just after clicking Fillet003 inside of Body named helic7 and I wanted to restore it. The Fillet003 was invalidated by a small dimensional change driven by Len1, Len3, Len5, Len7 defined in the Variables Spreadsheet.
Beside of this crash (which is the most critical bug), I think that this document evidences also a problem with the TopoNaming algorithm. In fact I have observed that Fillet003 is quite always invalidated by a change of those variables, This is strange because  Len1, Len3, Len5, Len7  modify only with the helices heights and should not change the Brep topology.

luzpaz, 2022-07-06:
Clicking on Fillet003 within helic7 results in:  
```
<PartDesignGui> Utils.cpp(1492): missing element reference in RX_HelicFilter_crashing#Fillet003.Base: ;#21a2;:Heff,E.Edge41
<PartDesignGui> Utils.cpp(1503): guess element reference in RX_HelicFilter_crashing#Fillet003.Base: ;#21a2;:Heff,E.Edge41 -> #21a2;:M;FUS;:Heff:7,E
<PartDesignGui> Utils.cpp(1503): guess element reference in RX_HelicFilter_crashing#Fillet003.Base: ;#21a2;:Heff,E.Edge41 -> #21a2;:M2;FUS;:Heff:8,E
7.9e-08 <App> Document.cpp(4293): Failed to recompute RX_HelicFilter_crashing#Fillet003: BRep_API: command not done
```

```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.706.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.07.06-edge)
Hash: 88fe682a2f892b9161134f57fcafad9511a64007
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)
```

CC @oficinerobotica @MisterMakerNL @bambukouk @snelweg  
Can y'all reproduce ?

bambukouk, 2022-07-06:
Sorry, traveling at the moment with no access to FreeCAD.
Will try as soon as I am back home.

oficinerobotica, 2022-07-07:
I tried the example file and did not get the crash. Strange because 2 people before me were able to reproduce.
However I was not able to fix the Fillet003 in any way. I tried the "toggle mode" removing all the edges to be filleted but there is something weird with the "Show preview" mode. I need to test this with a previous release. I got something similar with one of my files on the [2022.07.04] release.
[image]

realthunder, 2022-07-07:
Please run your application through terminal and paste the crash call stack printed in the console.

Regarding the fillet failure, it is because your sketch of the Pad before the Fillet is not created properly, resulting slight misalignment of the handle to the spring coil when you change size. And OCC fillet algorithm can't handle this kind of misalignment. See the video below on how to fix it.

https://user-images.githubusercontent.com/1207888/177697022-b12d9665-aa2d-48d3-aa70-12926479e330.mp4

MisterMakerNL, 2022-07-07:
I get a 
Illegal storage warning when opening the fillet and when I change it too 0.1 it crashes to desktop.
[image]

realthunder, 2022-07-07:
I can confirm the crash, and it is due to some OCC bug. To avoid the crash, you must recompile OCC yourself with cmake option `-DBUILD_RELEASE_DISABLE_EXCEPTIONS=Off`. Conda build won't crash because it has a custom built OCC with this option disabled. I'll try to rebuild Windows release with upstream libpack which has the same tweak.

wsteffe, 2022-07-07:
Hi RT, thanks for build info and for hint on the misalignment problem.

closed 2022-07-07 by wsteffe