realthunder/FreeCAD #475: Path workbench won't load.

MisterMakerNL, 2022-07-04:
As soon as I got the message there is a new release I downloaded it.
Found out that path workbench didn't work.

`ModuleNotFoundError: No module named 'pivy'`

`OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 2022.704.26244 +4971 (Git)
Build type: Release
Branch: LinkDaily
Hash: a87320cc64699a8e4226bad2e9b37ac3a1296ab9
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
`

shandyba, 2022-07-06:
Hi, same `pivy` not found error is thrown by several workbenches for me as well, using MacOs Stable Conda build, running on M1 Pro.

Had to switch to BigSur Daily for now, which worked.

luzpaz, 2022-07-06:
Can't reproduce on Snap edge:  
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.706.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.07.06-edge)
Hash: 88fe682a2f892b9161134f57fcafad9511a64007
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)
```

MisterMakerNL, 2022-07-10:
Seems to be fixed in:
`OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
`

closed 2022-07-10 by MisterMakerNL