realthunder/FreeCAD #477: Undisired file focus change on TreeView object visibility toggle.

oficinerobotica, 2022-07-05:
When toggling visibility of object in the tree, either by clicking the eye icon or pressing the spacebar , FreeCAD will change focus to another file.

https://user-images.githubusercontent.com/26909793/177277894-00d6ce52-dcd7-42f5-9fe1-95a3c5452d51.mp4

OS: KDE neon User - 5.24 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.704.26244 +4971 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: a87320cc64699a8e4226bad2e9b37ac3a1296ab9
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

realthunder, 2022-07-09:
Are there anyone else getting this? I can't reproduce it despite using the same settings.

realthunder, 2022-07-09:
Possibly related to #487. I know this is a recurring problem on MacOS for some reason I am yet to find out. But Linux doesn't usually have this problem. Maybe I can get some clue if I can reproduce it somehow.

eblanshey, 2023-07-23:
@realthunder I am getting the same issue on Windows. There is nothing printed in the Report panel. Let me know how I can help debug.