realthunder/FreeCAD #478: Make a Sketch with only a point.

Neubo, 2022-07-06:
Hello,

I would like to make object like those ones : 

[image]

(They have been done with Freecad 0.21, in which the explained issue described futher does not occur, obviously)

To achieve this, I will use an Additive Loft, and I need a scketch with only one point (plus a base sketch: rectangle, circle ..)
I know the trick that one point in a sketch is always a contruction geometry, and when I try to toogle it, I have an error message: 

**<Part> ViewProviderExt.cpp(2181): Failed to compute Inventor representation for the shape of Unnamed#Sketch: Bnd_Box is void**

If I try to force the additive loft selecting the 2 sketch, another error then:

**1e-07 <App> Document.cpp(4293): Failed to recompute Unnamed#Loft: Loft: Sections need to have the same amount of inner wires as the base section**

Thanks in advance for the solution !

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.704.26244 +4971 (Git)
Build type: Release
Branch: LinkDaily
Hash: a87320cc64699a8e4226bad2e9b37ac3a1296ab9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: French/France (fr_FR)

realthunder, 2022-07-09:
The above commit should fix the error, but not sure if that's all. Can you please provide a sample file showing the problem?

Shkolik, 2022-07-09:
Issue still exists in:

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)

Sample file:
[sketch_point_loft.zip](https://github.com/realthunder/FreeCAD/files/9077589/sketch_point_loft.zip)

There are 2 issues:
1) Sketch that has only 1 point (not construction) still not visible in 3D view 
2) Unable to loft from circle and point

Hope that will help.

Neubo, 2022-07-11:
Hello,

I confirm, it doesn't work.
Just let me know if you need more details, like a video.

nice week everyone !

Neubo, 2022-08-30:
Hello,

I've tested the Linux version (snap) and it's fixed there.
Any chance to get a rebuild of the Windows version ?

Thanks !

Neubo, 2022-09-26:
Hello @realthunder ,

I hardly represent myself the amount of work to release/compil the PC version, but it would be nice to have a refresh.
Or if someone could explain me how I could generate it myself ?

Many thanks !

realthunder, 2022-09-26:
It is not an easy task if you haven't tried before. Got busy at work these few days, but I'll try make one soon.

Neubo, 2022-09-26:
Thank you very much !

discip, 2022-10-05:
@Neubo
Since the initial issue is fixed, please consider closing this issue. 😊

Shkolik, 2022-10-05:
> @Neubo Since the initial issue is fixed, please consider closing this issue. 😊

It's not confirmed on Windows yet - windows version is 0709 still...

Neubo, 2022-10-06:
Hello, I agree with @Shkolik , in a past issue (https://github.com/realthunder/FreeCAD_assembly3/issues/866) it was solved in the Linux Version, but not in Windows' one.
That's why I leaved this issue opened. Am I right ?

Neubo, 2022-10-25:
Hello !

I've tested with the last Windows version, and the "Additive Loft" in "Part Design" works with a sketch with only one point.
But unfortunately, it doesn't with the "Loft" function in "Part" workshop :

[image]

This works in Freecad 0.21 (FreeCAD_weekly-builds-30694-2022-10-20-conda-Windows-x86_64-py310).

Just ask if you need more information / screenshot / video / logs.
Thanks !

Neubo, 2022-11-20:
Hello !

The Loft function (in PART workbench) still doesn't work in freecad FreeCAD-asm3-Daily-Win64-LibPack-Py3.8-20221112.
Error is 17:58:27  <Exception> TopoShapeEx.cpp(1539): Profile shape is not a vertex, edge, wire nor face.

LEt me know if you need more information.

julia-m-stephenson, 2023-01-12:
I am seeing the same issue under windows.

My Linkstage version

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United Kingdom (en_GB)

Working vanilla version
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.4.27
[/code]

After the loft on vanilla you see this.
[image]

With Linkstage I get same error as OP
[image]

Here is a simple test file
[loft_parts_vanilla_linkstage.zip](https://github.com/realthunder/FreeCAD/files/10404232/loft_parts_vanilla_linkstage.zip)

realthunder, 2023-01-12:
I am currently doing merge with upstream which already have this feature. Hopefully I can get it out this month.

Neubo, 2023-07-31:
Hello,

Juste tested with last tip version (merged with upstream).
Still doesn't work (make a loft with a point in Part workshop)

13:34:42  <Exception> TopoShapeEx.cpp(1631): Profile shape is not a vertex, edge, wire nor face.
13:34:42  Recompute failed!

[image]

```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2023.731.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 311c0066f7d2e5599a76337706627b6c27f6dc2b
Python 3.11.4, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * DVPScaled.brep
  * Manipulator 1.5.0
  * parts_library
  * QuickMeasure 2022.10.28
```

Neubo, 2023-10-19:
Hello @realthunder ,

I've tried with last TIP release, but it still doesn't work (thanks by the way, Draft Workbench is much better ;) )
One question from a neophyte : why did you use upstream version 2023.05.22 and not the last stable (0.21) released in August to make the merge ?

thanks !

Neubo, 2023-10-25:
Hello,

I confirm that it works not (realease 20231024) !

Thanks !

commit fa2127d0 (2023-10-22): Part: fix compound support when making loft

Fixes #478
closed 2023-10-25 by Neubo