realthunder/FreeCAD #485: Add option to use Hold Right-Click and move mouse to rotate camera

ChaosBlades, 2022-07-07:
Currently the right click is only used to change navigation style. It has been extremely frustrating using FreeCAD when right click is not camera rotation since that is how I use Fusion 360, TinkercAD, Ultimaker Cura, and PrusaSlicer. All the modeling tools I use have the below camera movement hotkeys except FreeCAD so every time I try to use FreeCAD I have to unlearn muscle memory.

Right-Click to rotate
Middle Mouse button to pan
Scroll Wheel to zoom

ChaosBlades, 2022-08-29:
I was about to close this since FreeCAD 0.20 implemented TinkerCAD navigation which is what I wanted but looks like the latest version of RealThunder FreeCAD still has not merged that code.

https://github.com/FreeCAD/FreeCAD/commit/ef100d55e9d50bd74ed8cc505a5d059de02cc146
https://github.com/FreeCAD/FreeCAD/commit/549e5b56502518d41865465d1e0953e12b10b4d8

duarteframos, 2023-01-17:
@realthunder Any chance we can get TinkerCad navigation style merged back into your branch?
It has been available in official FreeCAD releases since 0.20, but unfortunately it hasn't yet found its way here.

BloodyRain2k, 2023-04-10:
Maybe someone who knows how to could make a pull request?

duarteframos, 2023-04-10:
I suspect Realthunder is probably quite busy with the Topological Naming Problem development and merging it to official master.
His hands are probably full and this is likely low priority right now.