realthunder/FreeCAD #489: element axes rendered incorrectly when transformed

m1pro, 2022-07-11:
OS: macOS 12.4
Word size of FreeCAD: 64-bit
Version: 0.20.31151 (Git)
Build type: Release
Branch: 2022.07.06-edge
Hash: 1dd47fd7ecdcd8d5f776ced0a702a7a7dc5d4ee5
Python version: 3.8.13
Qt version: 5.15.4
Coin version: 4.0.0
OCC version: 7.4.0

Element axes render issue when setting any offset. Only x-axis (red) if affected.
See e.g. setting z position value:

[image]

closed 2022-07-11 by realthunder

realthunder, 2022-07-15:
The commit hash shown in your about information is incorrect. Please see the referenced commit above. Did you manually checkout some commit? Because tag 2022.07.12-edge is pointing to neither commits. Or, you can simply use the `HEAD` of my branch.

m1pro, 2022-07-16:
Actually tag 2022.07.12 points to the right commits: 
https://github.com/realthunder/FreeCAD/compare/2022.07.09-edge...2022.07.12-edge

I've checked locally and there was a problem on my side because the the mentioned file did not update.
Repeated the process and now all seems to be correct.

commit 7bb856b1 (2022-07-11): Renderer: fix handling of SoAutoZoomTranslation

Fixes #489
@m1pro