realthunder/FreeCAD #491: [feature request] Hide nav cube with shortcut.

MisterMakerNL, 2022-07-11:
So Solidworks has this nav-cube that only gets shown when you press space twice.
Which is pretty nice since you don't really need the nav-cube all that much.
Also we can already make the nav-cube bigger so why not make it huge and then only use it when you hit a key.
If it would hide automatically after you clicked on it would be a nice bonus but not necessary.  
Example on how it works in Solidworks:
https://www.youtube.com/watch?t=23&v=Lq4LeClPrGM&feature=youtu.be
I know you should be doing merge stuff, but this is way more fun. :)

discip, 2022-10-05:
@MisterMakerNL
1. There is already an option to autohide the nav-cube.

2. Additionally, you could configure a specific pie menu containing all the views you like.

MisterMakerNL, 2022-10-06:
Could you explain this auto hide feature?

discip, 2022-10-06:
@MisterMakerNL
Just click the little box on the bottom right of the _nav-cube_ and find the entry **`Auto hide`**.
Hover over it and you will find 2 entries that can be ticked.
If you check **`Navigation cube`**, the _nav-cube_ will disappear as soon as you move the mouse away.