realthunder/FreeCAD #496: polyline Modifier did not work in newest version

yahbluez, 2022-07-12:
The M modifier for polylines is not available in the version from 9.7.2022

realthunder, 2022-07-13:
Upstream has refactored keyboard shortcuts for Sketch workbench. You can assign your own using `Tools -> Customize -> Keyboard`

[image]

yahbluez, 2022-07-14:
The shortcut is still set to G,M but didn't work anymore. I can select a polyline but i can not change the behavior with G,M. In versions before 709 it worked as expected.

Shkolik, 2022-07-15:
Works fine with G,M on:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git)
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)

yahbluez, 2022-07-17:
2022.716 the bug is gone!

In two previous version even when i changed the key it didn't work.

On the same system the vanilla version(s) did not show that bug and older RT version ether.

yahbluez, 2022-07-17:
.

closed 2022-08-21 by yahbluez