realthunder/FreeCAD #497: Fillet crashes the aplication (segmentation fault)

rafacho, 2022-07-12:
I am trying to apply a fillet like the image below, but when I change the radius, all the application crashes.
[image]

The error reported by the terminal is:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f09a8cff090]
#1  0x7f099fc0dd74 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, opencascade::handle<Geom_Surface> const&, TopLoc_Location const&, double&, double&, bool*) from /snap/freecad-realthunder/55/usr/lib/libTKBRep.so.7+0x174
#2  0x7f099fc0de52 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, TopoDS_Face const&, double&, double&, bool*) from /snap/freecad-realthunder/55/usr/lib/libTKBRep.so.7+0xb2
#3  0x7f099fc1c21b in BRepAdaptor_Curve2d::Initialize(TopoDS_Edge const&, TopoDS_Face const&) from /snap/freecad-realthunder/55/usr/lib/libTKBRep.so.7+0xdb
#4  0x7f09a1a99f36 in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_Curve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_Surface>&, opencascade::handle<BRepAdaptor_Curve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const from /snap/freecad-realthunder/55/usr/lib/libTKFillet.so.7+0xd76
#5  0x7f09a1a9f5d8 in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) from /snap/freecad-realthunder/55/usr/lib/libTKFillet.so.7+0x778
#6  0x7f09a1aa1f18 in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) from /snap/freecad-realthunder/55/usr/lib/libTKFillet.so.7+0x108
#7  0x7f09a1aa6b19 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) from /snap/freecad-realthunder/55/usr/lib/libTKFillet.so.7+0x69
#8  0x7f09a1a6c8dd in ChFi3d_Builder::Compute() from /snap/freecad-realthunder/55/usr/lib/libTKFillet.so.7+0x42d
#9  0x7f09a1bfcd41 in BRepFilletAPI_MakeFillet::Build(Message_ProgressRange const&) from /snap/freecad-realthunder/55/usr/lib/libTKFillet.so.7+0x31
#10  0x7f09a05663cd in BRepBuilderAPI_MakeShape::Shape() from /snap/freecad-realthunder/55/usr/lib/libTKTopAlgo.so.7+0x8d
#11  0x7f09a2eaa39d in Part::TopoShape::makEShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /snap/freecad-realthunder/55/usr/lib/Part.so+0x5d
#12  0x7f09a2eaa484 in Part::TopoShape::makEShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /snap/freecad-realthunder/55/usr/lib/Part.so+0x84
#13  0x7f09a2eab835 in Part::TopoShape::makEFillet(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, double, double, char const*) from /snap/freecad-realthunder/55/usr/lib/Part.so+0x115
#14  0x7f09521cbea5 in PartDesign::Fillet::execute() from /snap/freecad-realthunder/55/usr/lib/_PartDesign.so+0x205
#15  0x7f09aafcd882 in App::DocumentObject::recompute() from /snap/freecad-realthunder/55/usr/lib/libFreeCADApp.so+0xa2
#16  0x7f09a2c9041d in Part::Feature::recompute() from /snap/freecad-realthunder/55/usr/lib/Part.so+0xd
#17  0x7f09521392fc in PartDesign::Feature::recompute() from /snap/freecad-realthunder/55/usr/lib/_PartDesign.so+0xdc
#18  0x7f09aaf42319 in App::Document::_recomputeFeature(App::DocumentObject*) from /snap/freecad-realthunder/55/usr/lib/libFreeCADApp.so+0x1e9
#19  0x7f09aaf5a143 in App::Document::recomputeFeature(App::DocumentObject*, bool) from /snap/freecad-realthunder/55/usr/lib/libFreeCADApp.so+0x73
#20  0x7f0951b700b2 in PartDesignGui::TaskDressUpParameters::recompute() from /snap/freecad-realthunder/55/usr/lib/PartDesignGui.so+0x22
#21  /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2ec1f0) [0x7f09a9b701f0]
#22  0x7f09ac04c797 in Gui::QuantitySpinBox::valueChanged(double) from /snap/freecad-realthunder/55/usr/lib/libFreeCADGui.so+0x47
#23  0x7f09ac04d373 in Gui::QuantitySpinBox::updateFromCache(bool) from /snap/freecad-realthunder/55/usr/lib/libFreeCADGui.so+0xb3
#24  0x7f09ac050740 in Gui::QuantitySpinBox::focusOutEvent(QFocusEvent*) from /snap/freecad-realthunder/55/usr/lib/libFreeCADGui.so+0x90
#25  0x7f09aa6fd46d in QWidget::event(QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x9fd
#26  0x7f09aa86992a in QAbstractSpinBox::event(QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xaa
#27  0x7f09aa6b9dc3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#28  0x7f09aa6c2bb8 in QApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a8
#29  0x7f09abaa1b18 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/55/usr/lib/libFreeCADGui.so+0x88
#30  0x7f09a9b3919a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12a
#31  0x7f09aa6c037c in QApplicationPrivate::setFocusWidget(QWidget*, Qt::FocusReason) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16c
#32  0x7f09aa6f6f0f in QWidget::setFocus(Qt::FocusReason) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x12f
#33  0x7f09aa6f7b91 in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x141
#34  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#35  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#36  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#37  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#38  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#39  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#40  0x7f09aa85eb97 in QScrollArea::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x17
#41  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#42  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#43  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#44  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#45  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#46  0x7f09aa6f7a9e in QWidget::focusNextPrevChild(bool) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4e
#47  0x7f09aa6fda8e in QWidget::event(QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x101e
#48  0x7f09aa86992a in QAbstractSpinBox::event(QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xaa
#49  0x7f09aa6b9dc3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#50  0x7f09aa6c3366 in QApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x956
#51  0x7f09abaa1b18 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/55/usr/lib/libFreeCADGui.so+0x88
#52  0x7f09a9b3919a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12a
[... cut ...]
#64  0x7f09a9b37c9b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#65  0x7f09a9b3fe44 in QCoreApplication::exec() from /snap/freecad-realthunder/55/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#66  0x7f09ab9f1e79 in Gui::Application::runApplication() from /snap/freecad-realthunder/55/usr/lib/libFreeCADGui.so+0x3e9
#67  /snap/freecad-realthunder/55/usr/bin/FreeCADLink(+0x4ed0) [0x558a35fb3ed0]
#68  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f09a8ce0083]
#69  /snap/freecad-realthunder/55/usr/bin/FreeCADLink(+0x52ce) [0x558a35fb42ce]
```

realthunder, 2022-07-12:
Can you please provide a sample file to reproduce the crash?

rafacho, 2022-07-13:
This is the part I am working on and the bug happens.
[https://drive.google.com/file/d/1bzSDBjNCN27s4eoebJcNrugt42W35naz/view?usp=sharing](https://drive.google.com/file/d/1bzSDBjNCN27s4eoebJcNrugt42W35naz/view?usp=sharing)
When I put the fillet and change it to 3mm the application crashes.

luzpaz, 2022-07-15:
@rafacho can you please always post the full [about](https://wiki.freecad.org/About) info with each ticket you post? Thanks!

rafacho, 2022-07-15:
Sorry @luzpaz, here follows:
```
OS: Ubuntu Core 20 (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 2022.712.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.07.12-edge)
Hash: 8ce366738c70dde14b276f68f6e2f6a380eac004
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)
```

BePeMee, 2022-09-01:
I'm experiencing the same issue:

https://user-images.githubusercontent.com/48593752/187967780-688b7cd4-a9e0-47b6-87cf-9cb42d30c36e.mp4

When I select the edges on the other side, it does not crash.
[samplefile.zip](https://github.com/realthunder/FreeCAD/files/9471879/samplefile.zip)

Version info:
OS: Linux Mint 21 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.709.28846 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c880c050c5f7601e4bae6740de71500edf2be0c7
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)