realthunder/FreeCAD #501: dragging Part Feature into an Empty Body not working any more

wsteffe, 2022-07-14:
Up to a few days ago I was able to drag a Part Feature (in example a Mirror) into an Empty Body.
May be I am doing it wrongly. I tried also to apply ctrl or ctrl+alt but not able to do it.
is anybody able to drag  Body (Mirror #1) into Body2 in the file [test.zip](https://github.com/realthunder/FreeCAD/files/9109161/test.zip) ?
[image]

master-puller, 2022-07-14:
@wsteffe Check if you have changed any settings. When I drag the mirror into Body 2, I get a reference BaseFeature.
[image]

wsteffe, 2022-07-14:
@[master-puller](https://github.com/master-puller)[master-puller](https://github.com/master-puller)
You image shows exactly what I wanted to have. 
The only setting I changed recently is "Allow cross coordinate links .." in Options/General/Objects.  Just reverted it but still not able to do what you did. 
When I am trying to drag Mirror into Body2, FC shows me a "no entry sign".
Have you synced recently with linkdaily ? Because I was able to do that up to the last sync/rebuild (3 days ago)

master-puller, 2022-07-14:
I am on 7/09 version, Win10 x64, currently don't have a script to compile FreeCAD, so i'm not on the bleeding edge.

commit 8f542573 (2022-07-15): Part: fix ViewProvider::canDropObject()

Fixes #501
@wsteffe
closed 2022-07-15 by realthunder