realthunder/FreeCAD #503: Display Style Settings Overwritten

master-puller, 2022-07-14:
When I attempt to change the display styles for selection to disable transparent overlay of object, settings do not save and are restored to default. This happens only on the `Display > Draw Styles > Hidden Lines` submenu.
I am attempting to turn off transparency on selection, yet this issue does not allow it. Video below of the bug in action

https://user-images.githubusercontent.com/19511301/179030915-30a1bda6-7a1b-4635-bf07-7b7e60f5b97a.mov

master-puller, 2022-07-14:
I found the solution to my previous problem, but the bug is still there which blocks those settings.

realthunder, 2022-07-14:
The option you want to turn off is called `Selection on top`. It is a selection behavior option, not relevant to draw styles. You can use shortcut `V, T` to toggle the option.

[image]

master-puller, 2022-07-15:
So, Even with Selection on top disabled, you cannot apply any of the Hidden Line override colors

realthunder, 2022-07-16:
Hidden Line override is only applicable if you set the draw style to hidden line, but your screen recording shows you are using the shadow draw style.

closed 2022-07-16 by master-puller