realthunder/FreeCAD #506: Freecad up to 20220721 missing Icons in Snap

ckvsoft, 2022-07-22:
Hi
[image]

In the Freecad Stage 3 Link versions up to 2022.07.21 various icons are missing in the snap image. Could someone please fix this? See photo.
The options on the left don't work either. These are always grayed out

ppd, 2022-07-22:
Disclaimer: I can't reproduce the issue. It smells like an icon theme is in use that does not have those icons present or that does not have a snapped version available, so some standard theme is used that also does not have those icons.

There was this change recently: https://github.com/realthunder/FreeCAD/commit/d3be8ee894f2f2d00c235069665366e212cbcd6e

> [...]so FreeCAD uses its included icons instead of the system icon theme on Linux
> Fixes missing/changed icons after installing new workbenches from
> the addon manager.

I'd try changing the parameter BaseApp/Preferences/Bitmaps/Theme/ThemeSearchPaths to `true` (or false, if it is already true).

As for the unclickable sidebar items: That's much nicer in the upstream snap (see screenshot). 
[image]

It seems like nothing "clickable" gets added in the LinkStage snap. Cherry-picking whatever commit is responsible for this from upstream would really improve the UX.

ckvsoft, 2022-07-22:
> Did this just happen recently? Did Previous version have this problem?

I had that in previous versions. Reported it now because nothing has changed up to the current version. The app images work, but I have other problems there

ckvsoft, 2022-07-22:
> Disclaimer: I can't reproduce the issue. It smells like an icon theme is in use that does not have those icons present or that does not have a snapped version available, so some standard theme is used that also does not have those icons.
> 
> There was this change recently: [d3be8ee](https://github.com/realthunder/FreeCAD/commit/d3be8ee894f2f2d00c235069665366e212cbcd6e)
> 
> > [...]so FreeCAD uses its included icons instead of the system icon theme on Linux
> > Fixes missing/changed icons after installing new workbenches from
> > the addon manager.
> 
> I'd try changing the parameter BaseApp/Preferences/Bitmaps/Theme/ThemeSearchPaths to `true` (or false if it is already true).
> 
> As for the unclickable sidebar items: That's much nicer in the upstream snap (see screenshot).

I've tried all sorts of themes. No change. I'll try that with Theme Search Paths.

What or Where is the Upstream Snap?

ppd, 2022-07-22:
> What or Where is the Upstream Snap?

https://snapcraft.io/freecad

However, those nice bookmarks in the sidebar of the dialog are only present on the edge channel. There's also an edge/toponaming channel for a preview of @realthunder's integration work of some of the stuff that's in this repo here.

ckvsoft, 2022-07-22:
Ok, thought that was a different one.
I am using sudo snap install freecad --edge.

ppd, 2022-07-22:
> I am using sudo snap install freecad --edge.

So this has the same issue for you?

ckvsoft, 2022-07-22:
Yes, that's exactly the version I made the report about. Just thought you meant a different Snap version

ckvsoft, 2022-07-22:
I now have this installed on my company computer. It's exactly the same there. So it should be because of the SNAP version.

I am using Kubuntu 22.04 in dark theme

ppd, 2022-07-22:
If you'd be so kind as to report this issue  at https://github.com/freeCAD/freeCAD-snap then, that'd be nice.

ckvsoft, 2022-07-22:
Yes, exactly. I'm using the FreeCAD Link Branch version. And this one has the error. I got these with sudo snap install freecad --edge. Did I ask in the wrong git?

ckvsoft, 2022-07-22:
Sorry
Now that got me a bit confused too.
Of course I have the version with me

sudo snap install freecad-realthunder --edge

Installed. But it's the same in both versions

ckvsoft, 2022-07-22:
I think the problem is with Kubuntu.
I changed the KDE theme from Breeze to Fusion and it works. However, none of the other themes work.
DontUseNativeDialog also has no effect. The same dialog is always displayed.

ppd, 2022-07-22:
> DontUseNativeDialog also has no effect. The same dialog is always displayed.

I think that only works in the freecad snap. @realthunder disabled it in freecad-realthunder permanently.

ckvsoft, 2022-07-22:
I change the Exec part in the Desktop file from
`Exec=env BAMF_DESKTOP_FILE_HINT=/var/lib/snapd/desktop/applications/freecad-realthunder_freecad-realthunder.desktop /snap/bin/freecad-realthunder
`
to
`Exec=env QT_STYLE_OVERRIDE=Fusion  BAMF_DESKTOP_FILE_HINT=/var/lib/snapd/desktop/applications/freecad-realthunder_freecad-realthunder.desktop /snap/bin/freecad-realthunder
`
This works, at least until the next update

realthunder, 2022-07-22:
You can still bring back the system file dialog by holding `Shift` while clicking the button. Note that in snap the system file dialog has trouble opening a file with external linked documents.

ppd, 2022-07-22:
> Note that in snap the system file dialog has trouble opening a file with external linked documents.

But only if you use a host system with an old xdg desktop portal. At some point in the future, the vast majority will have a file chooser that does not proxy local files.

ckvsoft, 2022-07-23:
> You can still bring back the system file dialog by holding `Shift` while clicking the button. Note that in snap the system file dialog has trouble opening a file with external linked documents.

Nice, but it works for me now and I can live well with the fusion theme

ppd, 2022-07-23:
@ckvsoft you can always copy the desktop file to your home directory. In this way, the changes will never be overwritten by updates.

ckvsoft, 2022-07-23:
@ppd  Yes that is actually true, thx

ckvsoft, 2022-07-24:
Of course, this only works if I open it via the desktop link. When I say open file with... not. I still don't have a solution for this. Of course, if I set the theme to Fusion, it already works.